opentx/opentx #6059: [Companion 2.2.2.RC5] Simulator creates .luac files

LapinFou, 2018-07-26:
Hi folks,

Even if the compilation option **luac** is not enabled, the simulator will always create a **.luac** file.
Then this file will be copy during the SD card synchronization process.
I tested it on Win10.

mpaperno, 2018-07-26:
Compilation options do not affect simulator (unfortunately), that's only built with a fixed set of options per radio type (unless you build your own, of course). 

There should not be any negative impact from having the .luac files on SD card even if radio doesn't use them (except the wasted disk space).  

In the next version of SD sync (2.3) you can specify a file pattern to exclude, so one could add `*.luac` to the exclusions if one wishes.

Only serious bugs will be fixed in 2.2.2 at this point, it's almost at release stage.

Please re-open if anyone can think of a better way to handle this or if I'm missing some bigger issue here.

closed 2018-07-26 by mpaperno

LapinFou, 2018-07-27:
Maybe the synchronization process could check if the **luac** option is ticked or not?
If not, then **.luac** shall be ignore.
😄

mpaperno, 2018-07-27:
Oh now you're getting fancy!  :stuck_out_tongue: 

That seems like a good idea though.  My only concern is excluding stuff w/out the user knowing about it (even if it is probably for their own good :) ).  We could easily tack `*.luac` onto the end of the Exclusion filter, though if it's hidden (the "extra options" aren't shown) then it wouldn't be obvious.  But maybe that's OK/good enough?

See screenshot at #6009

mpaperno, 2018-07-27:
Also I was going to make all the sync options persistent (saved to app settings)... so it would be a question of if the auto-appended `*.luac` should be saved to settings or not... and if not (would be my vote) then how to distinguish it from a `*.luac` that the user actually added themselves.

3djc, 2018-07-28:
Yeah trying to be to second guess users intention always kinda bites you back ! If a user doesn't want luac, then he should actively remove them using the future settings, imho

LapinFou, 2018-07-29:
My concern is for the users who don't tick the luac option, but does use the simulator. Some .luac files will be created by the simulator, then synchronized...
I didn't try, but what will append if the user (which does have an "unwanted" .luac file) does modifiy/update a LUA script. What the radio will use? The new .lua or the old .luac file generated by the simulator?

LapinFou, 2018-07-29:
Thanks Kilrah. This is good to know.
How the check is done? By checking the "modify date" attribute?

3djc, 2018-07-29:
luac are not even recognized as scripts if luac option is not there

mpaperno, 2018-07-29:
> By checking the "modify date" attribute?

[Yes.](https://github.com/opentx/opentx/blob/2.2/radio/src/lua/interface.cpp#L461)

mpaperno, 2018-07-29:
Actually another scenario is that a `.luac` file name is "hard-coded" somewhere in a script (which loads other scripts). The firmware will still load the .luac file even if `luac` is disabled (`luac` technically only enables the pre-compiler feature and ability to select .luac files in the UI). In fact the file can be called anything at all and the Lua processor will still try to load it, which was the only way to use pre-compiled scripts in the past.  

Also the same issue would happen if user had specifically selected a .luac file as a script in the UI while `luac` was enabled (or modified the settings in some other nefarious way :) ).  In this case the .lua version probably doesn't even exist.

So I think that negates auto-excluding .luac.

LapinFou, 2018-07-30:
> So I think that negates auto-excluding .luac.
I agree.
According the explanations of @3djc & @kilrah , having some "useless" .luac files generated by the simulator will have not impact at all on the correct behavior.
Thanks a lot for your time. :)