opentx/opentx #6060: Bricked X12S after upgrading firmware with Horus Bootloader (USB)

Zboubie, 2018-07-27:
Tryed 3 or 4 times to flash firmware 2.2.2RC5 with differents compilation options, and all goes well.
But finally after flashing from USB, bootloader get stuck, need to disconnect battery, but radio doesn't restart at all...

steps to reproduce:
1-Upgrade firmware with USB
2-upgrade completed
3-Eject USB device (got Windows's sound)
4-Bootload stuck with message "USB Connected"
5-Tryed to shutdown radio by long press on power button
6-Disconnect battery
7-Reconnect battery
8-Unabled to start OpenTX; with or without SD Card, even in bootloader mode (T5+T6+Power button)
9-Connect radio to PC: not detected, no sound signal for USB device

youtube: https://youtu.be/joLQXwgw4vk

Brand new Horus X12S FCC :'-(
Windows 7
Open TX2.2.2 RC5
Companion: 2.2.2 RC5
Previous firmware: opentx-x12s-multimodule-lua-luac-fr-2.2.2RC5.bin
Upgraded firmware: opentx-x12s-ppmus-multimodule-lua-luac-fr-2.2.2RC5.bin

3djc, 2018-07-27:
While it is clear you do have an issue, we do not believe at this point it is related to an OpenTx bug.  There are a number of external things, from Horus hardware defect, USB cable issue, ... that could lead to that.

What you could do is connect your Horus unpowered (without battery)  to your PC and check STM32 bootloader (do not assume that your driver is still installed, Windows wipes it regulary during updates).  If you have it, then you can recover your radio. If not, it probably needs to be serviced (unless someone else has a better idea, I'm no Horus expert !)

Zboubie, 2018-07-27:
I finally succeed to reinstall firmware by unpluging battery and connecting to PC. But it requires several attemps before Windows correctly detects the STM32 Bootloader (maybe I didn't waited enough after unpluging the battery).
So this is definitively not an hardware issue, but a bug during update. 
The update procedure was correctly made, but It will probably be hard to reproduce/fix this bug...

Zboubie, 2018-07-27:
damn, tryed again, to start in bootloader mode, Horus stuck without uploading firmware...