opentx/opentx #6061: A Toggle switch (on)-off-(on) as trimm funktion for X9D X9E at hardware setting

hrenz, 2018-07-28:
X9D and X9E have no free trimm switches  as X10, X12 with Trm5, Trm6 have

But I can replace a switch or install one more with a toggle switch (on)-off-(on)

At Hardware setting for X9D, X9E it would be fine to have  a selection as a Trimm-Switch
with same funktion as normal trimmswitches have, store value, autoInc and autoDec

bitogre, 2019-01-01:
I would also like this.  I find having many trim switches to be very nice. Especially for setting up Aileron differential and other in-flight adjustments of Global Variables (Special function "Adjust GV" with Source being a trim). However, 4 on my Taranis X9D plus does not seem to be enough. I notice that the Horus radios had extra Trims (Trm5 and Trm6) and I am looking to modify my Taranis to have similar functionality.

Currently, if I set the "Adjust GV" with Source to a 3 position switch, the GV will be -100, 0, or 100. It will not increment. I could get the same functionality that I am looking for my creating two special functions (and two logical switches if I want to be able to enable/disable the special function): one for incrementing and another for decrementing the GV but that is a bit more effort and I have to return the switch to center before incrementing or decrementing again. When doing the same thing with a trim, it only takes one line in the special functions page and holding the switch in the up or down position eventually results in continuous increment/decrement of the GV.

I am looking to replace one or more 3 position switch (like SB, SC, or SG) with a switch that looks the same but is momentary in the up and down position (spring loaded to return to the center off position). Then I want to be able to configure that switch in the Radio "Hardware" page as a "Trim" (make the options be "Toggle", "2POS", "3POS", and "Trim"). Once the switch is configured as a "Trim" in should have most, if not all the functionality of a trim (though the only functionality I care about is that of "Adjust GV" described above).

kilrah, 2019-01-02:
It would be really complex to make this runtime-selectable, and would waste a lot of resources for anyone not using it, so I don't see it happening.

Should be able to use a channel for that purpose.

bitogre, 2019-01-02:
@kilrah, how would we be able to do this with a channel?  Did you mean it could be done with the GV and not a channel?

Also, if you read my comment (or my version in #6280 which I closed when I found this), I think can be done at runtime without wasting a lot of resources by not giving full trim functionality (no need for it to be an input) but just the same behavior as a trim for purposes of the special function "Adjust GV" with Source being a trim.

kilrah, 2019-01-02:
See on CH5 here. 
https://www.dropbox.com/s/lokwqjigzqvr3kd/document1.otx?dl=1

hrenz, 2019-01-02:
We can select at hardware 2Pos and 3Pos switch and have one singel side SH with momentary  Funktion.
But we habe not a auto repeat funktion or auto inc/dec like a trimm switch has.
We can replace a normal 3Pos switch with 3Pos Switch with momentary Funktion (on)-off-(on)
So it would be nice to have autorepeat funktion like a trim switch has.

bitogre, 2019-01-02:
I fully agree with @hrenz's latest comment.  I think where things are getting confused is the use of the word "trim" but I am not sure what would be a better word to use.  I used "trim" because that is the existing auto-repeat behavior (and not a new input) that we want available for other 3 position switches.

bitogre, 2019-01-03:
> See on CH5 here. You could even adjust the speed with a pot.
> https://www.dropbox.com/s/lokwqjigzqvr3kd/document1.otx?dl=1

Looked at this and it is a nice trick.  Thanks for showing it to me.  I might have use for it in the future.

But still not what we want with regards to adjusting GV.  Yes, I know you can use a special function to load a channel value into a GV but I see problems with getting it to work properly when each Flight Mode has its own GV Value to adjust.  How many more logic switches and special functions are needed to use this trick to adjust Flight Mode GV and have them properly load when switching Flight Modes?

kilrah, 2019-01-03:
Just one, the value that's adjusted by an Adjust GV SF is always the one for the currently active flight mode.

bitogre, 2019-01-03:
> The value that's adjusted by an Adjust GV SF is always the one for the currently active flight mode.

Yes, and that is the problem.  If we hold the switch up or down, why can the increment/decrement not repeat the way it does with trims?  And why require us to create 2 special functions (one for increment and one for decrement) when you only need one to do it with trims?

> You may need some mixer lines to set the channel to the GV's current value on FM change.

I just checked I cannot figure out how to load the channel with a GV.  I can see how to do it with a Special Functions but it looks like I would have to create many logical functions to do the detection of the change of Flight Mode (a number equal to two times the number of Flight Modes minus one).  So, for my DLG that has 6 Flight Modes, I would need 11 Logical switches for Flight Mode Change detection, 3 Logical Switches for this trick, a extra channel, and 2 Special Functions to read/write the GV to/from the channel.  That seems very excessive.  It gets worse if adjusting multiple GVs like this.  If this change is implements as desired, only 1 Special Function is needed per GV to adjust.

kilrah, 2019-01-03:
Thing is it would have to be done as a selectable option, because some of the original purposes that requested the Increment/Decrement GV specifically require it not to autorepeat.

> And why require us to create 2 special functions (one for increment and one for decrement) when you only need one to do it with trims?

There is very specific and somewhat convoluted behind the scenes behavior when the source for AdjustGV is a trim.

bitogre, 2019-01-03:
> There is very specific and somewhat convoluted behind the scenes behavior when the source for AdjustGV is a trim.

Ok, so use a different mechanism that is not convoluted.  We used the word "trim" to describe the behavior, not to describe how to implement the behavior.

bitogre, 2019-01-03:
I was looking at the code and saw that isRepeatDelayElapsed is used to repeat playing items.  Why not use a similar mechanism (could re-use lastFunctionTime in CustomFunctionsContext to store when last repeated) but with a shorter repeat value for this?  Maybe add argument to isRepeatDelayElapsed to indicate scale of repeat value.  Just a suggestion.

closed 2019-11-15 by stale[bot]