opentx/opentx #6062: Cannot set correct GPS time zone X9D+

arnabraha, 2018-07-28:
Unable to set +5:30 GPS time zone. Can set +5 or +4, but not +5:30 Hrs.

This is a very useful feature with "Adjust RTC" function. I believe its about time to fix this years old problem.

In Radio Settings
[image]

**General information**
 - OpenTX version 2.2.2
 - Hardware Frsky X9D+
 - OpenTX Companion 2.2.2

arnabraha, 2018-07-28:
I would like to clarify more - The time zone in 30min steps cannot be done with the Radio and the companion software both; since this feature does not exist. It is very important to address this issue - with OpenTX being adopted at so many countries with various time zones.

3djc, 2018-07-28:
Duplicate #1653

As it requires an eeprom change, it wont be done before next major version

closed 2018-07-28 by 3djc