opentx/opentx #6063: Modelselection does not remember the actual model / modelcategory

ParkerEde, 2018-07-29:
Hi,
since 2.2.2:
while open modelselectionscreen, der Horus X10 marks always the first model in the first modelcategory.
This is not good.
In 2.2.1 the modelslectionscreen is coming up with the last selected model in the last modelcategory. This much more better than the behavior in 2.2.2. 
Can you please fix it?

Kind regards
Ralf

ParkerEde, 2018-07-29:
I've to specify the behavior in 2.2.2.

new information:
while open modelselectionscreen, the  Horus X10 marks always the first model in the last active modelcategory which was selected at PowerOn. 2.2.2 does not store the model/modelcategory while the radio is powered on and you change to a model from another modelcategory.

tueddelclick, 2018-08-27:
It's the same for me:
I switch on the HorusX10 (openTX 2.2.2) and open the model selection. It displays the models from the category "Trainer". Now I am switching to the category "Glider" and choose a model. Opening the model selection screen again the Horus is again displaying the category "Trainer" (the category that was active at the transmitter start). In version 2.2.1 it behaves not like this. This version noticed the category of the last chosen model correctly.

ParkerEde, 2019-08-17:
It would be nice if this request could be reconsidered.

tueddelclick, 2019-08-18:
Because it is working in 2.2.1 I hope there‘s a chance to solve it.

pascallanger, 2020-03-10:
This weird behavior is still in the current 2.3.5.

3djc, 2020-03-10:
Obviously, if not, this would be closed !

pascallanger, 2020-03-10:
Just preventing the bot to close it ;-)

commit 6609481c (2020-06-08): Another attempt to fix #6063 (#7706)
closed 2020-06-08 by bsongis