opentx/opentx #6066: Trim Reverse

drummaker, 2018-07-31:
I sometimes get aileron direction screwed up in my head while in the work space.   Generally I have limits and Offset on the outputs, and 3 rate settings,   When this happens, and I go out to the field and realize my error. It would take A LOT of programming to change the offsets and max rates on the outputs and anything like this done in the mixes..    When this happens I always just switch the input to negative instead of positive percentages.    This then creates a scenario where I must use the trim in the opposite direction.... Usually I discover this while air born   It sure would be nice to either have a checkbox to reverse trim... or have it automatically reverse.

hrenz, 2018-08-05:
yes an old problem, an open issue since years
we can change to negativ Inputs but can not reverse the trim
this in inconsequent, could  be solved simple

lshems, 2018-08-05:
There are lots of cases where this logic is NOT applicable. It would ask for a separate trim inverse to be able to choose.

You can add trims manually with a separate second mixer line. Assign weight as you wish.

Finally, reversing weight to accommodate mixing up of left and right aileron servo connections is not neccesary in opentx.

Just change the two sets of mixer lines.

There is no fixed order in opentx.

3djc, 2018-08-09:
There is a clear, known rule for inversion. Inverting in input does not apply to trims, inverting in mixer does. This rule cannot be changed as it would affect all existing model (even if we wanted to, which we probably don't)

closed 2018-08-09 by 3djc