opentx/opentx #6068: 2.2.2 X9E always shows "Model still powered" in last screen

FS4000, 2018-07-31:
After updating from 2.2.1 to 2.2.2, I turn off my X9E (confirmation selected)
and in the last screen there is always the message "Model still powered"
without any receiver is in the near.
If a model is still powered, the right message in **bold letters** is shown before.

Greets and thanks for your hard work, Uwe

3djc, 2018-07-31:
I cannot reproduce. Please give more details

FS4000, 2018-07-31:
I selected the German language for my Opentx. Maybe its only in the German translation.

Greets, Uwe

closed 2018-08-07 by 3djc

OpenMik, 2019-07-20:
Also for my radio 9XE with OpenTx 2.2.4 message
"model still powered" is showed every time during shutdown
Italian language

kilrah, 2019-07-20:
Have you selected the "shutdownconfirm" option?

OpenMik, 2019-07-20:
yes

kilrah, 2019-07-20:
Then it's jsut the wrong message but the requirement for pressing is what you asked for.

OpenMik, 2019-07-20:
Yes is only a wrong message
At shutdown appears together the the "Model still powered" message and the shutdown confirmation but  pressing  enter the radio shutdown correctly