opentx/opentx #6070: Horus X12S - momentary switch F

Stuntman2, 2018-07-31:
I have replaced the one way switch F with a momentary switch and using the sticky function in Logical Switches it acts as a pull on pull off switch. Up to the final release (2.2.2) companion simulator mimicked the transmitter ie when TX first switched on SW-F was off both on the TX and in Companion simulator, after upgrading to 2.2.2 the simulator outputs screen showed SW-F as on, but on the transmitter at switch on it is off.

kilrah, 2018-07-31:
You should set the correct switch type in the Hardware settings instead of the sticky function...

closed 2018-07-31 by kilrah

expipiplus1, 2019-02-19:
@kilrah Would you be able to post a link to one of these tutorials, please?