opentx/opentx #6074: Multi-Point Curve Variables

jlorenco, 2018-07-31:
The ability to assign a variable to individual points on a multi-point curve, that act like a GV on the entire curve, would be a very useful feature.  This would allow in-flight tuning of of individual points on a curve for things like mixes for multi-point curves which is currently not possible.  Applications include complex sailplane mixes that require time consuming flight testing to dial in because the tuning must be done by trial and error on each point, landing after each try.  Also, throttle and pitch curve tuning for helicopters and quads among others. 

One way around this was suggested by another user to use logical switches to divide the throttle travel into increments that match the range of points on the curve, then adjust a GV (one for each point) and store the GV values.  After flight, the values can be transferred to the actual curve.  No a very elegant solution.

The ability to assign a control/switch to each variable for points on the curve, makes adjustment of these curves a simple and useful feature, much like the GV works now on the global weight.

3djc, 2018-08-01:
Since you can setup curve with LUA, I would suggest you look into making somethiing like that with LUA, to tailor it to your unique needs

jlorenco, 2018-08-01:
The whole point of this feature request is that this is not something for a unique need. This feature could be used with any curve to provide the ability to adjust flight characteristics real time. Having to write a LUA script every time you want to have the ability to do real time adjust of a curve point is not very practical.  I believe this was the whole point of having GV's for use in mixes, etc. This is an expansion of that philosophy to make the system more capable without the need for extensive programming.

kilrah, 2018-08-02:
It wouldm't need a different script every time -  a single "curve adjuster" script could work for every scenario and be shared.

jlorenco, 2018-08-02:
Perhaps, but having to add a Lua script for each time you want to add an adjustment is not exactly user friendly.  This is a feature that other radios already include such as Multiplex and Graupner to name a couple. I also just found that this same feature was previously requested in Issue #1433 and is still open.  It doesn't seem to me that adding useful features should be ignored just because they can be done with a Lua script.  The main function of the radio is to fly aircraft and making that better/easier should be the goal, not making programming more complicated.  That said, I realize there are some features that are extremely difficult to incorporate and if this is the reason, then that is understood.  If not, it could be a big improvement.

kilrah, 2018-08-02:
Please propose a more user friendly native way then. Won't be that easy to do better IMO.
Also... with Lua it can be done today with probably a day'a worth of work. Doing it natively would likely require pretty deep changes, aka would need to be done for a major version, and not the next one because of time constraints (unless someone else contributed it), so wouldn't get in before 6-18 months or so.

jlorenco, 2018-08-02:
I realize this is significant and wouldn't be a quick addition, but I think it is a worthwhile one to add when possible.  The idea is that there would be way to select a variable (to support control) in the curve menu for Y values (and potentially X values), instead of just fixed values.  The same as a GV can be selected instead of a fixed value in other items such as mixes.  Selecting the variable box would allow the value for that point to be assigned to a new variable (or I supposed existing GVs could be used to simplify the process).  Then it would work just like a GV does now for that point (which may be a reason to use GVs for this purpose).  I think this would be something users are already familiar with and be intuitive to use.

mpaperno, 2018-08-02:
Simply use GVs and multiply each point by the value of a GV.  You can already assign controls to change GV values, and set the step size and limits.  Though with a multi-point curve seems like you'd need a lot of knobs either way... or adjust multiple points with one GV.  And granted this is limited to the 9 GVs available, but it would probably makes sense to expand that, if anything.

3djc, 2018-08-03:
You also have to keep in mind that in firmware option, nocurve and nogvar are among the most demanded options, so your feature suggestion isn't exactly mainstream (which mmeans on the priority list, this will have a hard time making it). 

On the other end, you can design an lua that would read current curve, let you know by voice while flight how many point it has, use a switch (trim ?) to move from point to point, voice you the current value of that point, use anoother trim or knob or whater to adjust, go to next point, or whatever you feel like...

jlorenco, 2018-08-03:
I guess it's matter of considering where FrSky is headed with the radio line such as the Horus radios.  These seem to be targeted more towards pilots who prefer to use a radio to fly rather than spend time programming.  These can't even run all the scripts like telemetry scripts, so are not targeted at the more traditional OpenTx users.  Maybe improvements to the FrOS operating system is a better approach?

kilrah, 2018-08-04:
Horus radios also run OpenTX, just not by default. A lot of people who want to do advanced stuff (I would think adjusting curve in flight is "advanced", very few people and especially not those who want to "fly and not spend time programming" would bother) on them do use OpenTX.

lshems, 2018-08-05:
@jlorenco

Opentx is not aiming at being a user-friendly radio operating system.

This may be annoying to lots of people, but is as it is.

In general, if something can be done in some way, the developer don't see much use in implementing it.

FrOS should be aiming at user friendliness, but then will be limited just because of that.

Perhaps the entire opentx GUI will some day be written in Lua. Then nothing is to crazy to add, and nobody depends on some server having to build their versions of screens and functions.

Funny thing is that the servo by servo editing page on the XLite is functionally an exact copy of the page I wrote in Lua for my own GUI. So we are not that far of perhaps.

schwabe, 2018-08-05:
I am not to which page you refer but most of the xlite UI is from the x7 UI which in turn is from the old 9x UI. Those UI is older than any capabilities.

And you confused user friendly with beginner friendly.

lshems, 2018-08-06:
I'm not confusing anything. Just different opinions.

notlebowski, 2018-08-06:
For the OP I wrote a setup a week ago that's working. 
I monitored his ticket seeing that he's referred to LUA doing that.
Together with my solution I described a "Special Function" that could change the values in curves which, if implemented, could be used to manipulate the curves during a flight.

I have always been able to do whatever I wanted without using LUA so I never did anything with that. Because I was intrigued by his problems I looked into the LUA-solution and came up with this code.

It should do the thing the OP asks for (possibly needs some more improvements), but it alters nothing to the curve.
Can someone point out what I'm doing wrong?
It doesn't have any syntax errors, it just doesn't change the curve although I think I'm altering it.

It is able to fetch the curve, calculates the zones in which the Y-value falls at that moment.
It increments the y value and it writes the curve with **model.setCurve( c - 1, curve )**, but that doesn't seem to work.

**INPUTS**
I3:Thr Thr Weight(+100%)
I5:CMP CH3 Weight(+100%) Curve(CV1)

**Mixers**
CH3 I3:Thr Weight(+100%)
CH5 I5:CMP Weight(+100%)
CH6 luascript

**CV1**
Standard, [-100, -77, -56, -21, 0, 36, 61, 70, 100]

Input=CH3
Curve=CV1
Switch=SA

local inputs = {
                 { "Input", SOURCE },
                 { "Curve", VALUE, 1, 16, 1 },
                 { "Switch", SOURCE },
               }
			   
local outputs = { "curvm" }
			   
local function init()
end			   
			   
local function run(i, c, en )
  local input
  local zonesize
  local zone
  local curve
  
  if en == 0 then
    return i
  end
  -- Read Curve c
  curve = model.getCurve( c - 1 )
  if curve.points < 5 then
    return i
  end
  
  zonesize = 2048 / (curve.points - 1)
  -- input is shifted to 0~2048 for easier calculations
  input = i + 1024
  zone =  math.floor( (input / zonesize) + 0.5 )
  
  if en > 0 then
    if curve.y[zone] < 100 then
	  curve.y[zone] = curve.y[zone] + 1
      model.setCurve( c - 1, curve )
	end
  else
    if curve.y[zone] > -100 then
	  curve.y[zone] = curve.y[zone] - 1
      model.setCurve( c - 1, curve )
	end
  end
  return i
end

return { input=inputs, output=outputs, run=run, init=init }

notlebowski, 2018-08-06:
I now know why it doesn't work.... 
model.setCurve has a complete different structure than model.getCurve
pfff.... I picked a nice first project to learn lua

jlorenco, 2018-08-06:
You're doing great!  Can you also share how you accomplished this without using Lua?

notlebowski, 2018-08-06:
I have it working now....

> Can you also share how you accomplished this without using Lua?
I was the one giving you an example with gvars.
You said it worked

local inputs = {
                 { "Input", SOURCE },
                 { "Curve", VALUE, 1, 16, 1 },
                 { "Switch", SOURCE },
               }
			   
local outputs = { "curvm" }
			   
local function init()
end

local function run(i, c, en )
  local input
  local zonesize
  local zone
  local curve
  local converted_curve
  local y
  
  if en == 0 then
    return i
  end
  -- Read Curve c
  curve = model.getCurve( c - 1 )
  if curve.points < 5 then
    return i
  end
  
  zonesize = 2048 / (curve.points - 1)
  -- input is shifted to 0~2048 for easier calculations
  input = i + 1024
  zone =  math.floor( (input / zonesize) + 0.5 )
  
  if en > 0 then
    if curve.y[zone] < 100 then
	  curve.y[zone] = curve.y[zone] + 1
	end
  else
    if curve.y[zone] > -100 then
	  curve.y[zone] = curve.y[zone] - 1
	end
  end
  curvey={}
  for i=1, curve.points do
    curvey[i] = curve.y[i-1]
  end
  
  model.setCurve( c - 1, {smooth=curve.smooth, y=curvey} )
  return i
end

return { input=inputs, output=outputs, run=run, init=init }

jlorenco, 2018-08-06:
Yes, I have it working but only stores curve values to GVARs, it doesn't actually move the points on the curve.  Glad to see you have this automated version working and thanks for sharing it.

notlebowski, 2018-08-06:
This one actually moves the curve points
It is better to put the script in /SCRIPTS/FUNCTIONS and use an SF to call it

SF1  !SA- - Play Script (curvem)

Set the LUA-script in the radio and give it the 3 parameters CH3, CV1 and SA
Fill CV1 with a curve with at least 5 points
You then don't need all those logical switches anymore and neither the GVARS
Just have an extra INPUT with CH3 as source and CV1 as curve

closed 2019-11-15 by stale[bot]