opentx/opentx #6078: Horus x12S

PatrizioPasquali, 2018-08-02:
Hi
When i use the monitor function, after few seconds, the Horus display stop working;  freezing the image and i can't switch off the tx; to do this i must unplug the battery.
I loaded the last relased version of Opentx 2.2.2 on my Horus X12S .
I choosed LUA and LUAC options
Take into account that i am very new with Horus, and presently I am traying to set up some display widgets, so some my wrong can be.

thanks
Patrizio

mpaperno, 2018-08-02:
The "monitor function" is the channels monitor screen?

Does this happen without touching any controls on the radio? Specifically the scroll wheel?

You should probably post your .otx file.

PatrizioPasquali, 2018-08-02:
[document 2 aug 2018.zip](https://github.com/opentx/opentx/files/2253147/document.2.aug.2018.zip)

antgeebee, 2018-09-27:
Im having the same issues except if I hold the off button down for around 10 sec or so it switches off...until 
the next time the monitor function is selected and the rotary wheel rotated
nothing but the servos and the sound works

bsongis, 2019-09-10:
Anybody succeeds to reproduce it?

closed 2020-07-08 by stale[bot]