opentx/opentx #6081: Channel output values rounded/skipped

Ultrawipf, 2018-08-03:
This might not exactly classify as a bug or might be known and "unfixable".
Came across some noticable rounding while working on the usb portion a while back and investigated why the intervals don't look quite right.

The reason why it bothers me is that the transmitted resolution is not exactly even 11 bits even when the analog input resolution is already sampled with 12 bits.

**To Reproduce**
Steps to reproduce the behavior:
1. Fresh model without changed mixes or offsets, calibrated sticks.
2. Use a test lua script with something like getValue('ch1')
3. Move channel stick very slightly from the edge
4. Initially value jumps from -1024 to -1022 instead of -1023, some numbers are skipped, then the steps are 1 again for some time. Clear signs of premature rounding somewhere in the chain.

**Expected behavior**
As the analog inputs are sampled in a higher resolution no further noticable rounding should take place in the output if no complex scaling or offsets are introduced in the mixer. I expect no values in the output to be skipped here and every value in the full 11 bit range should be possible.

**General information**
Taranis X9D+
OpenTX 2.2.2

kilrah, 2018-08-03:
Calibration. Stick usable range is never the full ADC range, and the mapping will cause aliasing.

Ultrawipf, 2018-08-07:
Sure, the sticks don't use the full range, but the adc is 12 bit and you can see that on the debug screen. Between the mapped steps are many raw adc counts. The adc resolution is more than enough.

After thinking about that for a bit i am sure the resolution **can** be improved to mitigate the issue, but at the moment its probably not worth the effort and might collide with jitter filtering and mixer mappings. After all the sbus protocol does support 11 bits and performance and resolution improvements are always a good idea if some day the input mappings are revisited. 

But feel free to close the issue if there are no plans. I am currently very happy with openTX as it is.

schwabe, 2018-08-07:
Usable range from min to max is more like 1000. if you go only from 1V-2V you only have about 1400-2700 which are about 10 bits.

closed 2019-11-15 by stale[bot]