opentx/opentx #6086: OpenTx Companion 2.2.2 Bug in Model Comparison for Horus X10

Hartinger, 2018-08-05:
The model compare feature in Companion 2.2.2 (Horus X10 profile) seems to have a bug with regards to the switch warnings. See attached screen shot showing actual settings in model editor vs. report from model compare facility (German language).

[image]

I could not observe this phenomenon when the Taranis QX7 profile was used.
I can't remember having seen this with 2.2.1 but cannot rule it out either.

elecpower, 2018-08-05:
Confirmed issue for Horus X10 and X12 profiles. X9D+, X9E, X7 and X-Lite profiles tested okay. All tests were performed on a new model file for each profile.
Repeatedly running the print function results in the switch values changing and duplicating switches.
[image]

elecpower, 2018-08-06:
So I performed some more tests on using Horus 10 profile and discovered the model print is affecting the model settings.

Steps to reproduce:
1. New Horus 10 profile
2. New model file
3. New model using wizard accepting all defaults
4. Open model edit  and select settings tab - noting all switches displaying in up positions
5. Close model edit
6. Click print model - noting different switch settings printed to step 4
7. Close print
8. Open model edit settings tab - noting switch settings changed.

I haven't been able to reproduce this behaviour for Taranis profile. A couple of screen shots for evidence.

[image]
[image]
[image]

PR #6100 (2018-08-29): Fixes Companion model print switches
Fixes #6086
closed 2018-08-29 by bsongis