opentx/opentx #6090: Add JavaScript support?

davidngrc, 2018-08-06:
OpenTx is written in C++
Lua script does not look like C++
Javascript is more like C++

Is it possible, implement JavaScript support in OpenTx?
so more people have "website programming still" can create more "plugin"?

so many open source c++ javascript parser
like https://github.com/gfwilliams/tiny-js

lshems, 2018-08-06:
If javascript does look that much as C++, why not add to the opentx source code directly?

Lua has a totally different objective (if I understood the developers well). It is for users that want to use a text editor to change minor details to shared LUA scripts for small tasks. Developers that can write C++ can write LUA scripts, I would reckon.

By the way, the differences between the radios is already a complicating factor to create an 'ecosystem' for LUA. Making a second script language available would comlicate things (or it would need to be a decision to make LUA obsolete).

Just my view on it.

schwabe, 2018-08-06:
Javascript and C++ have very little in common apart from the basic syntax. Especially embedded C++ and javascript. Also java script interpreters tend be very big. Lua on the other hand is designed to be a small interpreter that can be used a script language for embedded devices.

Normal Javascript engines don't fit the job needed here.

mpaperno, 2018-08-06:
Is it possible? Probably (what isn't :) )!  I even like the idea, would have maybe preferred that to Lua in the first place.  That TinyJS link led me to a fork of it which is even designed for embedded systems (STM32 even) and has graphics drawing functions and a lot of other extensions.  Neat.

That said, it would be a relatively major undertaking to write a new JS API... and this is totally glossing over any potential integration issues that will come up.  I bet it could eventually be more popular than Lua though.

@lshems JS is a lot more like Lua than C++ in terms of programming with it... it's a scripting language.  The syntax just happens to be somewhat similar to C/C++ (but only superficially).  It also has an object system which is very easily extensible (even right from within the scripts themselves) which makes it, in theory, quite suitable for the task at hand.  It's actually much easier to use than Lua in a lot of ways.

mpaperno, 2018-08-06:
> Lua on the other hand is designed to be a small interpreter that can be used a script language for embedded devices. 

Not really... IMHO.  It's been used as such by a few projects and there are some other projects which tried to make it more embed-friendly, but I don't think it was ever designed as such from the start.  Witness the numeric format "fiasco."  But it does give some consideration to "simpler" systems of yesteryear (which were less powerful than our STM32  :) ).

mpaperno, 2018-08-06:
As another example, Qt uses JS (ECMAScript, technically) to extend a lot of its C++ API into a [scripting environment](http://doc.qt.io/qt-5/qtqml-javascript-topic.html) they use in their [QML](http://doc.qt.io/qt-5/qtqml-index.html) declarative language.  All the scripting is in JS, but you can also use C++ objects, enums, structs, etc, including custom ones, as well as objects defined in the QML itself.  And QML is specifically targeted for mobile and embedded applications (this is full ES v5.1+, not some stripped-down version).  So, it's definitely doable and JS/ES is very extensible.

Personally I prefer a strongly-typed language, but I'm guessing that wouldn't be a very popular choice... :wink:

kilrah, 2018-08-07:
Or it would just reinforce the saying that "web developers are not real developers" :P 

(he did mention it was for people with "website programming skill")

davidngrc, 2018-08-10:
@kilrah I really think more people know javascript than LUA.
look at chrome, firefox plugin, and the QT that @mpaperno mention.

@3djc the existing LUA script is way less compare to the project that use javascript.
look at the jQuery plugin, so many plugin...
why? more people know javascript than LUA.

raphaelcoeffic, 2018-08-10:
Well, JS has been used by browser for ever, which is why. But frankly, anyone with the most basic notion of programming can learn LUA is minutes. The most time is spent with the APIs rather than the language syntax itself. Even with with JS in OTX, you would still have to familiarize yourself with the API.
It took me 5mins to get my first LUA scripts with OTX working.
AND, given the effort needed, I think we’d rather spend the time with other things. Anyway, if you want to have JS support, please make a PR with working code. Then I’ll look at it.

closed 2018-08-10 by 3djc

kilrah, 2018-08-10:
> look at the jQuery plugin, so many plugin...

Even if a small enough interpreter exists most of the plugins you'd want to use would likely exceed memory resources.

mpaperno, 2018-08-10:
jQuary... LOL.   I bet `XMLHttpRequest` won't work either...  :wink:

bsongis, 2018-08-10:
I would like Angular + MD on my radio :)