opentx/opentx #6093: .deb and PPA do not contain x-lite model.

meoso, 2018-08-09:
Using both Debian and Ubuntu, the provided packages do not contain the FrSky x-Lite model. (the packages themselves are 2017)

3djc, 2018-08-10:
Not sure where you are  downloading from but the .deb link on the official page is 2.2.2 : http://www.open-tx.org/2018/07/28/opentx-2.2.2

deb : https://downloads.open-tx.org/2.2/release/companion/linux/companion22_2.2.2_amd64.deb

The PPA need to be updated yes

meoso, 2018-08-10:
**yes, the link to the PPA is definitely outdated:
https://launchpad.net/~opentx-test/+archive/ubuntu/rel/+packages
`2.2.1` is the latest listed with date `2017-12-17`**

and indeed the .deb is up-to-date, i suppose it did not replace the ppa version properly when i tried it the first time.

OpenUAS, 2018-08-16:
PPA update would be really appreciated. Also to avoid having a new PPA for a new release.

OpenUAS, 2018-08-24:
@3djc No sweat, already appreciate that there is a PPA :+1:

benderl, 2018-12-05:
Any News about the PPA?

WuerfelDev, 2018-12-06:
Would be nice if there's a notice if on the downloads page as long as this is not up-to-date :)

(And also how do i even run this? After installing -the old version- i wasn't able to run it by command)

meoso, 2018-12-06:
`companion22` is the executable name, from the `.deb`

closed 2019-11-15 by stale[bot]