opentx/opentx #6094: Telemetry data offset- transfer to companion changes value

Squirlier, 2018-08-10:
Hello,

OpenTX companion: 2.2.2
 Horus x10S  X10-2.2.2

Telemetry Data in Horus x10s - offset saving ok.  (-10.00)
value after power cycle still -10.00

transfer to companion.. offset changed to -3.00  
transfer back to horus. value is still -3.00  

when changing value in companion to -10.00
and then transfer to horus. value is -10.00 OK

tested nothing else. 
maybe its ok when within range of 0 to -3.00 (not tested)

that was on id 0x0035 (for testing)
and also on id 0x0038

happens on new fresh model as also on existing model (3 days)

kilrah, 2018-08-25:
Cannot reproduce. Please specify all settings of that sensor or even better zip the MODELS and RADIO folders of the radio's SD card when set up in a way that would cause the issue when reading into companion and send that to us.

Squirlier, 2018-08-25:
Hello kilrah,
Thank you for testing.
The Problem exists if you set the Precision to 2 (0.00)
Then the offset limit seems to be -3.00 while transferring to companion.

Test:
on Horus x10S create new model.
cancel the wizard by pressing RTN.
go to model setup>telemetry
create sensor. 
name it aa
ID 0005 or 0025 or 0035 (maybe any number)
Precision  0.00       (  0.-- or 0.0 do work correct)
Offset -8.00     (like -80.0 or -800 in the other precision settings)
LOG ON.
EXIT Setup.
(Power cycle Horus)
Connect to PC
Get complete Model and transmitter settings (don´t know the Menu name)
open new model and check telemetry sensor ID00XX (see above)
Offset changed to -3.00

I hope i could deliver an understandable Test

kilrah, 2018-08-25:
Ah, precision to 2 was the key.

commit 9526f734 (2018-08-25): [Companion] Set decimals first on autospinbox or value can be truncated (fix #6094)
commit bf5e5874 (2018-08-25): [Companion] Set decimals first on autospinbox or value can be truncated (fix #6094)
closed 2018-08-25 by kilrah