opentx/opentx #6095: F.Req: Export telemetry Logs to VIRB Edit

skaman82, 2018-08-10:
It would be very awesome if we could do such overlays on our flight videos based on the opentx log files: https://www.youtube.com/watch?time_continue=19&v=2E572pFwTR8

3djc, 2018-08-10:
Well our logs are cvs, so ready to import in whatever tools needed,  not sure what more can be done from OpenTx perspective

skaman82, 2018-08-10:
There is a specific xml file format required for this (GPX files)

They look like this: 
<trkpt lat="-27.4107841" lon="153.2515232"><ele>44.785</ele><time>2015-10-04T02:08:09Z</time>
<speed>4.38</speed>
<hdop>1.0</hdop>
<sat>8</sat>
<gpsarPro:satsInView>0</gpsarPro:satsInView></trkpt>

skaman82, 2018-08-10:
https://developer.garmin.com/garmin-connect-api/sample-data/

mpaperno, 2018-08-10:
You can already export to KML (GoogleEarth format) from Companion.  If that overlay thing doesn't support KML natively (I'd be surprised), there are a number of converters out there which do.  You can even do it [online](http://www.gpsvisualizer.com/convert_input).

mpaperno, 2018-08-10:
$5000 license fee?   LOL!

closed 2018-08-10 by mpaperno

skaman82, 2018-08-10:
I don't think you will be needing a license for the GPX file format. I guess if it was easy to do, we have been seen some flight videos using this overlays already - so I guess its not so much "straight forward" for the public to do. To bad you are not interested

mpaperno, 2018-08-10:
You just upload your KML and you get a GPX file... what's hard about that?  Easier than figuring out Garmin licensing (or what they're even selling there), that's for sure.  Supporting GPX format is pretty much going backwards (KML is much richer).