opentx/opentx #6098: pls. make individual trims accessible by special functions and Lua

StefanKellerAC, 2018-08-12:
It would be fine to be able to set individual trims with the special functions and/or with Lua...

3djc, 2018-08-12:
Could you detail a bit what you have in mind (why, how, scenarios,...) ?

StefanKellerAC, 2018-08-12:
- instant trim could be explicit for only one channel
- presets could be set by events or switches or ...

- we could use it more flexible as a control
- I tried to make a cruise control for a car:
    trim on the channel is disabled. 
    with trim +/- a global Var is modified. which controls gas in the respective mIxer
    when I touch the stick the cruise control is disabled by a logical switch
    until here it is fine, 
    but then trim should be set to zero, 
    otherwise I have to start next time at the last speed which may be to high
    or when I use the trim button to increment the global var I will stuck at the end of the trim range

3djc, 2018-08-12:
I'm not sure I'm following you. You need to disable the trim in the current flight mode, and set 2 special function to inc / dec the Gvar, and one to reset it on a switch

StefanKellerAC, 2018-08-12:
yes, but then you will end using this in one direction when maximum trim is reached ( I have a X7)
... ok, I tried it again. Yes it works but I am confused by the permanent announcement "maximum trim reached "
if you go up and down you may never reach that point

-> setting the trim to a value would be much more flexible and elegant :-)

3djc, 2018-08-12:
There is not message about max or min trim reached if done as I said, (you probably did not set trim to '--' in your fight mode

kilrah, 2018-08-12:
You can already see trim values on the main page.

kilrah, 2018-08-14:
There are not shown by default but you can display them on a telemetry screen.

closed 2019-11-15 by stale[bot]