opentx/opentx #6101: X-Lite Switch Direction

macsleep, 2018-08-14:
I am using OpenTX 2.2.2 on an X-Lite. I have seen that Bertrand worked on the switch direction for the X-Lite in issue #5828. I still have to push e.g. switch A down (literally) to get a SA^ in OpenTX. Is this the correct behaviour or shouldn't this be the other way around?

3djc, 2018-08-14:
NO, it is right. There is a OpenTx rule that applies to all radio, that come in all king of shapes and forms.
The closest to your head is ### down, the farthest **up**

closed 2018-08-14 by 3djc

macsleep, 2018-08-14:
Thanks for explaining :-) .

3djc, 2018-08-15:
We understand it feels strange if you look at xlite only, given it's physical setup, but we also want to ensure it is consistent between radios (so you don't have one down on your x9d, while the same on xlite is up (or vice versa))