opentx/opentx #6102: Please add model name .wav file if any played when the model number in the select model list is hignlighted

racermark, 2018-08-15:
*Before suggesting a new feature here, please search the open issue if there is already a request for the feature*

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

I am still in need of having My Taranis on OTX play the model name file when selecting a new model as I highlight the new model number. What would work well I think would be after the number is highlighted for say 2 seconds the model name file would be played allowing me to select the next or scroll through several up or down without instantly playing each file.

This would be really helpful as I cant really read the screen well inside and not at all outside.
This may just be my last summer of flying as the only EYE I have left is going fast.

notlebowski, 2018-08-15:
I hope your request gets implemented in time.

lepicfoto, 2018-08-16:
It would be a mega cool feature yes a really cool feature 👍

PRGeno, 2018-08-16:
I think this would be a very useful and uniquely cool feature to be added to OpenTX.

closed 2019-11-15 by stale[bot]