opentx/opentx #6109: Comapnion Store/Retreive does not save and restore Global Functions

rdeanchurch, 2018-08-18:
On Windows10, Companion 2.2.2 and OpenTX 2.2.2 for a X7

Open and existing .otx with calib, trainer, hardware and Global Functions
Click
'Store calib. and hardware settings in selected profile'

Under File, NEW create a new .otx
 in the  new .otx click 
'Restore  calib. and hardware settings in from selected profile'
Look at the Global Functions tab
Notice 'Restor..."' does not replicate Global Functions of the original.

Is this deliberate or and oversight?

**Screenshots**
[image]

kilrah, 2018-08-18:
Global functions have nothing to do with hardware, they are user configuration. 
Hardware = stick calibration, battery type and settings, stick mode, volume, what switches are installed, what mods are done... things that may vary between 2 radios of the same type.

closed 2018-08-18 by kilrah

mpaperno, 2018-09-06:
Related: #4583