opentx/opentx #6113: Emergency mode (Notfallmodus) Horus X10S after normal flight (about 25 min)

Squirlier, 2018-08-19:
Open TX 2.2.2  (a5de713b) 2018-07-28
Horus X10S  LBT version
Non Eu option for D8 Mode possibility.
sd card is class10 16 gig intenso.

What happened:
After normal Flight with lots of speech sounds...
while still logging to sd card every 2 seconds...
Horus jumped in Emergencymode (or how it is called).
Receiver got wired and outputs started to shake a little... 
luck: landed already and motor didn´t selfstart while carrying plane. (good)

The weather was not too hot. 
I don´t know what happened.
Horus could be switched off and switched on. Emergency Mode was away.

If there are logs/files that can help i will provide them.

alex-vg, 2018-10-01:
Sounds and logging? Looks like a new occurance of #3771

kilrah, 2018-10-01:
As long as it is not repeatable and only occurring in rare occasions it's not a reason for concern. That's why emergency mode was developed in the first place.

closed 2018-10-01 by kilrah

scott-adamson1975, 2019-02-23:
This has just happend to me while in flight, lost control in a roll, running lua inav telemetry script, will try to retrieve logs, if any

Happened while doing a roll, what could cause emergency mode? Is a receiver brownout a possibility?

3djc, 2019-02-23:
Hard too say, if we knew we would just prevent it. It has nothing to do with receiver, it's on the radio itself. Likely suspect are LUA stuff, but it could be other things too. Logs are useless in this case

scott-adamson1975, 2019-02-23:
teh lua script may have just been about to warn me about low satelites and low voltage at teh same time, playing wavs