opentx/opentx #6115: Instant Trim will adjust a trim beyond the limit if it's already at or over the limit.

Miami-Mike, 2018-08-21:
As long as a trim setting has an absolute value < 250, meaning < 24.414%, the Instant Trims function will restrict the adjusted trim value to be less than or equal to that limit, unless the Extended Trims option is checked.

That's fine, but if the trim setting **is equal to or greater than the limit** then an activation with the stick deflected in the same direction will increase the setting even further. In fact, repeated activation of Instant Trim when a stick is held in a deflected position can increase the trim value as far as ±1000/1024 = ±97.656%, and this will happen even if the Extended Trims option is not checked.

This could happen if a user accidentally activates Instant Trim with the stick significantly far from center, and then activates it again. The trim could get so far off that a model could become uncontrollable, and it's not easy to get the trim back to where it belongs.

I discovered this in OpenTX Companion Simulator, Version 2.2.2.

kilrah, 2018-08-21:
That's intended so that if you do it once and it's not enough becasue things are so out of whack that standard trims aren't enough you can do it a 2nd time.

kilrah, 2018-08-22:
The Instant trim feature is great but definitely has to be used with care. Why exactly would you give full stick deflection then press the instant trim switch twice? It's far-fetched.
Far-fetched enough that nobody who actually uses the radio has ever reported doing that...

closed 2018-08-22 by kilrah