opentx/opentx #6119: Companion crashes when loading bitmap files in telemetry LUA scripts

zufus, 2018-08-22:
Hi,

I am testing a telemetry script for X9D+ that I found on the web. The script visualize the RSSI through a widget.

The script tuns just fine on the radio, but it hangs companion when I run it with the Telemetry Simulator enabled. The same script without calling the lcd.drawPixmap function runs perfectly. On the other hand, when I call the lcd.drawPixmap the script runs for few seconds, then companion crashes. Some times the script runs a little-bit longer, but it ends crashing anyway.

I am using companion on Debian GNU/Linux, 64 bit platform.

I am attaching here:

o) the model setup
[setups_scripts.zip](https://github.com/opentx/opentx/files/2311688/setups_scripts.zip)

o) the script (with the .bmp images)
[telemetry_script.zip](https://github.com/opentx/opentx/files/2311686/telemetry_script.zip)

o) the standard error output
[companion_output.zip](https://github.com/opentx/opentx/files/2311702/companion_output.zip)

I am available to provide further informations

Thanks

Marco

zufus, 2018-08-22:
Edit: the same script, with the same model setups on Windows runs just fine. So I conclude that this is a Linux-related issue, (please note that the bug occurso on both my Linux machines).

3djc, 2018-08-23:
It doesn't fail here under Debian Strech, Did let it run over 10 mins

mpaperno, 2018-08-23:
@zufus  
* Where did your installer/build come from, is it the official OTX release, or self-built?
* Which version are we talking about... 2.2.2 Release?
* Any chance this only happens with particular bitmaps (eg. `RSSIh08` but not `RSSIh01`)?

Thanks.

mpaperno, 2018-08-24:
Reproduced.  Mint 19 (Ubuntu/Debian based) which has Qt 5.9.5/gcc6 by default, using the QTX deb 2.2.2 release as well as versions built on the box, and also with Qt 5.7.1.  And, luckily, even debug builds.

Seems directly related to the copious amount of debug output from TRACE_SIMPRGMSPACE during file access.  More specifically to memory reallocation by QByteArray when buffering the debug output for display in Simulator.  Or even more specifically in `malloc()` of `libc-2.27` :)  

Looks like a recurrence of an ongoing issue I've dealt with before, last time due to the huge amount of debug output dumped during X10/X12 startups. That was "fixed" by increasing the buffer size IIRC, but apparently it was just a stopgap.

Possibly some Qts are more susceptible to this... eg. it crashes here quicker with 5.9.5 than 5.7.1 (all else being the same).  Last time I dealt/tested this was on 5.5.x IIRC, though I also remember 5.2 was very vulnerable.  Or it could be the compiler/libs version, or some combination of the above.

Anyway, I'm testing a fix now which avoids any re-allocations altogether (at the expense of reserving more memory up front), and so far it seems solid.

zufus, 2018-08-24:
well, good finding! :) Thanks!

Marco

zufus, 2018-08-30:
Dear Max, I confirm your version seems to fix the problem here, the script is now running since more than 10 minutes, and I restarted the simulator few times.

Thank you for your support, and sorry again for the time it took to reply :)

Marco

mpaperno, 2018-08-30:
Hi Marco.  That's no problem at all, thanks a lot for testing and confirming!  And again for the comprehensive reporting.  I will get a patch into the next version.

Cheers,
-Max

commit bb4909f5 (2019-12-06): Bsongis/mpaperno sim dbgbuff fixes (#7161)

* [Simulator][FIFOBufferDevice][FilteredTextBuffer]
  * Fix issues with dynamic memory allocation on Linux (closes #6119);
  * Fix occasional issue with long lines getting partially repeated in output;
  * Fixes filter buffer getting overwhelmed with some very verbose Horus startups (observed on Linux);
  * Adjust buffer max. size settings to int type (to match QByteArray size type);
  * Re-implement QIODevice bytesAvailable() and canReadLine() for efficiency;
  * Remove some unnecessary vars, other minor optimizations, cosmetics.

* [Simulator] Slig [... cut]
closed 2019-12-06 by bsongis