opentx/opentx #6121: Horus x10 model wizard 2.2.2

bny1, 2018-08-23:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version (e.g. 2.1.23 or N123)
 - Hardware (e.g. Frsky X9D+ or 9X Pro)
 - OpenTX Companion (e.g. 2.1.23 on Ubuntu 16.04)

bny1, 2018-08-23:
When you create a new model it just stay frozen on that page plus it dont show plane just the glider set up but wont go to the next step.

3djc, 2018-08-23:
Then you likely do not have thge right SD card content....

[image]

[image]

closed 2018-08-23 by 3djc

bny1, 2018-08-23:
were do i get the right content? i just downloaded 2.2.2

bny1, 2018-08-23:
it just the glider wizard but I just can get it to work

3djc, 2018-08-23:
Sorry but this is not a support site, please go to  community sites like rcforums or openrcforum for end user support