opentx/opentx #6123: It is possible to change UI page while range check is ON

kilrah, 2018-08-25:
On all radios

Allerhopp, 2018-08-27:
To be able to change page during range check was often useful for me, for example to watch the sensor screen in range check mode.

kilrah, 2018-08-27:
What do you want to see on the sensor screen during range check that is useful to said range check?

Allerhopp, 2018-08-27:
I am used to fly my rangetests/antennatests with a RTH quad and i watch distance, altitude, speed, heaading, RSSI, lost frames ....

kilrah, 2018-08-27:
OK...
But then think of the normal plane pilot who has the radio muted or volume set a bit low and does not notice he is still in range check mode after exiting back to the main screen (the reason for this is that it has happened).

It was thought until now that it was not possible and EXIT was the only way out to the main menu and home screen, thereby disabling range check. But it turns out you can change page, and when there you can exit and get back to the main menu with no indication of being in range check.

Allerhopp, 2018-08-27:
To be honest, i was surprised when i realised the first time it is possible at all. Safety first, i will work with the logfile in the future. lBtw. long page would also be helpful to be able to switch to telemetry screen/LUA telemetry screen, any chance to keep the long page function alive?

kilrah, 2018-08-27:
Huh? Long page is how you get to the telemetry screen from the home page. In the menus it is "previous page". No change there.

Allerhopp, 2018-08-27:
You are right of course, i had to press page, exit and long page.

mpaperno, 2018-09-09:
Having the RSSI displayed while bind is active seems like a nice feature (when it's available).  And/or perhaps a special screen (with big "BINDING" title) and a non-editable telem. sensors list.  

The little chirpy warning sound is easily missed.

mpaperno, 2018-09-09:
Ooops, I was reading "range" and thinking "bind".  It is also possible to switch pages during bind though... not sure if that ends binding mode (the chirps stop), but seems like an oversight.

kilrah, 2018-09-09:
Well that is much less of a problem since you can't fly while in binding mode...

mpjf01, 2018-09-09:
I reckon the "no change" position should apply to bind mode as well. For consistency and because we have all been present when someone has launched a model without checking first that he had control.  You will never stop that happening of course but it would be better to reduce the possible excuses.

PR #6124 (2018-09-18): Prevent page changes during value editing
To be discussed. Fixes #6123 but also affects all other value changes... which I think is actually a good thing, I have pressed PAGE by mistake during editing more than once...
closed 2018-09-18 by kilrah