opentx/opentx #6126: Flashing X12S with bootloader bricked radio

notlebowski, 2018-08-25:
This morning I used the bootloader to first add some FRK-files to the folder /FIRMWARE (only mentioning this, because it could be the cause) and then flashed "opentx-x12s-multimodule-noheli-lua-luac-en-2.2.2N667.bin"onto it.
I noticed a short pause during uploading.
After detaching the USB-cable the radio did not turn on again.

I am sure it can be fixed by removing the battery and flashing it again the classical way, but I don't have time for this now. Maybe tomorrow.
When I attach the cable to my X12S I'm not getting the "_STM32 bootloader_" in my device manager.
[image]
Windows 10 tells me that a USB-device is attached that's not recognized.

If I attach my X9D+ with the same cable I'm getting the "_STM32 bootloader_"

[image]

I don't know exactly which bootloader was on my Horus. I think it was running a post 2.2.2 firmware flashed in the classical way.

I flashed the X12S before, when the bootloader was just recently introduced.
It then also resulted in a non-function radio. I was however able to get a garbled screen and I was also able to turn the radio completely off so I could flash it again.

From the outside there is no sign the radio is turned on, but I think it's not completely turned off. That's why I can't flash it now in the traditional way before detaching the battery.

Is there something I can do to fix it without opening it?

notlebowski, 2018-08-25:
PS....
I just noticed that the backlight of the X12S is turned on

kilrah, 2018-08-25:
> I noticed a short pause during uploading.

Did you wait until the "Close" button got active, clicked it, then properly ejected the USB device?

> Is there something I can do to fix it without opening it?

Wait until the battery dies...

notlebowski, 2018-08-25:
Yes... I waited until the close button became clickable.

I just got home and noticed the backlight was still on. I don't even need it this weekend, so could wait a week even.

notlebowski, 2018-08-26:
This morning I noticed that the backlight was off, so I was able to flash my X12S using dfuutil
In total I used the bootloader for X12S 3 times.
1st time successful
2nd time it gave me a garbled screen...

waited 2 months

3rd time yesterday which gave me the status I described in this ticket.

I have been using nightlies for about 3 years and flashing with bootloader has always been reliable on the Taranis models (I vaguely remember I had a fail once too).

No regular user has had a need for the bootloader flash. This will happen on the release of 2.2.3
Only then will I know if my fails are an exception.
My Horus X12S is of the first batch.

notlebowski, 2018-08-26:
I probably unplugged it before shutting down because it needs to be unplugged before turning on.
I will do this in the future.

Is it an idea to unmount the drive after the flash has completed?

I never made a ticket for it, but I can remember that if you also do a write model in the same bootloader session those are not properly transferred.
I don't do that nowadays.

notlebowski, 2018-09-01:
This morning I gave the "flash with bootloader" another go.
After flashing I kept the USB-cable connected and tried to turn it off with the power button.
That's not possible (as I always thought).
But it was probably not what you ment.
This time I ejected the 2 drives before disconnecting the USB and turned it off.

The display goes off, but it never turns on again.
This time I also don't have the backlight turned on, so it will probably take days before the battery goes down and I can flash it again.
I'm using my X-Lite now, so I don't need it and if I really need it I will open it up and disconnect the battery.
I will not be using the bootloader anymore until something is changed in the code.

closed 2019-02-25 by notlebowski