opentx/opentx #6127: 2.2.2 is at least 30% slower accessing SD card than 2.2.1

teckel12, 2018-08-26:
*Before submitting,  search the open issue if there is already a report of the bug*

**Describe the bug**
2.2.2 (even the latest nightly build N668) is at least 30% slower with SD card access than 2.2.1.  So slow that it can cause huge lag in the UI.  You can see this easily with Lua scripts that do a lot of loading of scripts in the background loop.

**To Reproduce**
Steps to reproduce the behavior:
1. Download [Lua Telemetry 1.4.0](https://github.com/iNavFlight/LuaTelemetry/releases/tag/1.4.0)
2. [Install Lua Telemetry](https://github.com/iNavFlight/LuaTelemetry/wiki/Installation)
3. Run Lua Telemetry
4. Press `Menu` and scroll through menu (notice lag)
5. Use other menus in OpenTX (notice lag)
6. Downgrade to OpenTX 2.2.1 and repeat 3-5, notice diference

**General information**
 - Hardware (e.g. Frsky Q X7)

teckel12, 2018-08-26:
Video example: https://youtu.be/mPPkhgMAR0k

mpaperno, 2018-08-26:
What makes you sure it has anything to do with SD card access?  Have you done any testing with anything besides your one script?  Empirical statistics (where did you get 30%)?  I can't tell what's going on in your video, sorry.

I'm not saying you're wrong, but we're probably going to need a bit more to go on than "it seems slower when running my script."  I recall briefly looking at that script code recently and to be frank there seemed to be "room for improvement" in terms of efficiency.

teckel12, 2018-08-26:
@mpaperno I know it has to do with SD card access because I changed the script to not use as much SD card access (but perform the exact same action) and it became increasingly faster.

I then added code to the script to show the fps and then recorded what the fps difference was from 2.2.2 to 2.2.1 which is where I got the 30% slower.

It actually seems it's 30% slower everytime an additional script load happens. So if there's several, it's many times slower (as seen in the video example). To a point of being unusable and causing the entire UI to not be usable. This didn't happen in 2.2.1

This is enough to conclude that the issue is with 2.2.2 compared to 2.2.1 and report it here.

teckel12, 2018-08-26:
Efficiency and memory used is a constant problem with Lua. Previously, I went for execution speed. But in the latest release I was going for better memory usage. This was accomplished, but only when running 2.2.1 it seems ad something changed in 2
2.2 which makes loading scripts MUCH slower.

If you have suggestions for better Lua speed as well as lower memory usage I'd love to hear it. To date, any suggestions have not actually provided any better speed or memory usage (typically the opposite).

teckel12, 2018-08-26:
For further evidence, you can run v1.4.1 of Lua Telemetry compared with v1.4.0. You'll notice a HUGE difference in lag between these two versions. The only real difference is that 1.4.1 doesn't load as many scripts from the SD card. The functionality is essentially the same.

But, if you run opentx 2.2.1 the difference between Lua Telemetry 1.4.0 and 1.4.1 is not noticeable.

mpaperno, 2018-08-26:
There's no point in getting defensive or sidetracking about the general limitations of Lua on a tiny MCU like in the X7.  Impressive that it works at all.  If it works worse in 2.2.2 then that is the issue at hand (slowing down until "unusable" certainly doesn't sound good).  I was asking for more evidence to back the claim, and you have provided some, thank you.  Maybe this is enough to give one of the other devs some clue.

The best way to approach a problem like this is with a [minimum complete verifiable example](https://stackoverflow.com/help/mcve). Break it down to the simplest script which can demonstrate the problem -- don't ask busy developers to install some complicated system which is also hard to debug.  Be concise in descriptions.  Don't go back and add important information to posts because it's likely someone has already read it and will not see your edits (or at least mark them clearly if you do).

> I then added code to the script to show the fps and then recorded what the fps difference was from 2.2.2 to 2.2.1 which is where I got the 30% slower.

This is the kind of empirical evidence I would have mentioned up front.  Though w/out a concrete example we don't really know what you're measuring (not criticism, just fact).  Perhaps useful to put this into the MCVE I mention above.

> It actually seems it's 30% slower everytime an additional script load happens.

What does this mean?  Are you loading scripts inside each loop... and each loop is 30% slower than the previous?  Or that every additional script loaded at init will then slow down each subsequent run/background loop?

Thanks,
-Max

teckel12, 2018-08-26:
@mpaperno To be fair, there's hundreds of issues here and while I tried to search for a similar issue, I believed that someone had already posted the issue but in a different way where a search wouldn't easily find it.

So, I gave a quick snapshot of what was happening as well as how to reproduce it (even if a *bit* complicated, if you call coping a file and folder complicated). If it wasn't reported before, I would give more examples/evidence.  The sidetracking of the issue was started by you, suggesting that it was a coding problem on my end. Not sure if you've written complicated Lua scripts, but if you have I'm sure you've run into the problem with limited memory constraints. Which is why to someone without this experience would look at Lua code in a vacuum and conclude it's written poorly, suggesting that's the problem instead of opentx.

In any case, if we can focus on the incredible slowdown between 2.2.1 and 2.2.2 I would be more than happy to go that route. That was all I was doing till being derailed.

When I say it's like it's 30% slower for each SD access, I was loading a script inside a script and multiple scripts were loading for each loop. For each background loop, there would be maybe 4 scripts loaded. This resulted in the extreme slowdown as shown in that video.

I didn't experience this with 2.2.1. So I upgraded to 2.2.2 and then noticed the same slowdown. I then folded one of the loaded scripts into the parent script so it wouldn't need to load. This reduced the number of loaded scripts per background loop from 4 to 3. I noticed a speed up. So I added code to show the display fps and could confirm that it was faster.

I then folded the other two loaded scripts into the parent script and the speed increased to a point of being about as fast as 2.2.1. To get the 30% number, I added a single script load back and compared the fps difference.

In total, at first it was running at only about 3 fps with 2.2.2. After refactoring the code to not do any script loads, it was more like 18 fps. This significant difference doesn't happen with 2.2.1. 2.2.1 was obviously a bit slower with the script loads, but it didn't introduce any lag as a result.

As a result of this, the script now uses about 10k more of memory than it did with the loading of scripts technique used previously. Which is why it was doing these script loads, to greatly reduce the amount of memory used. That is, until 2.2.2 made this necessary programming technique unusable.

I'll look into this further and see if there's as easier way to reproduce it. But honestly, you'll still need to copy a file and a folder to reproduce it, which is exactly the same steps needed to install Lua Telemetry.

mpaperno, 2018-08-26:
>  The sidetracking of the issue was started by you, suggesting that it was a coding problem on my end

There is almost no code out there that doesn't have room for improvement (including my own). If my suggestion to revisit your code offended you, sorry.  But what you essentially posted originally is "my script now runs slower than it used to."  So yea, first suggestion is always "check/improve your script."

> if you call coping a file and folder complicated

There are 4 paragraphs of instructions. Starting with needing an iNav.  Honestly I didn't even read them all.  The least you could do is provide a ready-to-load OTX file with all the necessary sensors/etc.

Can you reproduce the slowdown issue in Simulator?  How about on another radio?  (most of us don't have X7).

> you'll still need to copy a file and a folder to reproduce it

You're missing a key point: Minimal example (eliminate everything not directly related to the problem -- this will also help you narrow it down whether you realize it now or not... you may even discover the real issue lies elsewhere).  And maybe another one: Make it as easy as possible for someone to help you.

> Not sure if you've written complicated Lua scripts

I wrote the OTX Lua pre-compiler feature and `loadScript()` to help work around the memory issue.  You're welcome.  So, yes, I have a pretty good clue about the limits and what low-impact Lua code looks like.  I think people tend to forget (or don't know what it means) that they're programming for a tiny embedded environment on a relatively slow single-core system, and then get upset when it runs slowly or doesn't fit in RAM.  Trigonometry functions, for example, are very expensive.  Disk access is slow.  Everything in a script which is not a comment or white space uses memory (including length of identifiers!).  Each numeric variable uses 64bits regardless of actual type.  And the scripts run in same process (thread/priority) as the GUI, so they will directly affect it and be affected by it.  And at a low priority so pretty much everything else on the radio will preempt the Lua tasks.

> multiple scripts were loading for each loop

I would strongly avoid loading anything from SD card inside run/background (or any rapidly repeating loop).  This is basically like putting a variable-length delay inside your loop.  With a slow or bad SD card it could easily fail altogether.  It may easily conflict with other files needing to be loaded at the same time (eg. voice alerts).  It may save some RAM (I wonder), but at the huge expense of slowing everything down due to disk access and reloading of Lua code.  Even pre-compiled Lua still needs to be interpreted.  Loaded script functions can be cached at init() instead.

I noticed you're using `loadScript()` with "T" flag... why? Relying on the fact that the plain-text version is not also in the same folder?  For debug purposes I'd set a local/global var and pass that to loadScript instead.

> The OpenTX team currently seems to be more interested in pointing blame than looking into the issue.

Nice one.  And ironic considering your first post here blamed OTX with no direct (stated/empirical) evidence.  Cheers.

teckel12, 2018-08-26:
@mpaperno I'll just focus on the problem at hand and ignore the noise...

As you don't need INAV or sensors or anything to duplicate the results in the video and see the lag, all that's needed to install the script is to copy the contents of the `dist` folder to your telemetry folder.  That's it.  You can download the script here: https://github.com/iNavFlight/LuaTelemetry/releases/download/1.4.0/LuaTelemetry.zip  Just open the ZIP, move the contents to your `dist` folder and run.  I'm sorry if the install instructions seemed daunting, they're designed for someone trying to setup telemetry from scratch, not just use Lua Telemetry.  In any case, to duplicate the results you don't need INAV, telemetry, or even a model at all, as it can duplicated without the model even being turned on as in the video example.

Dividing the script into multiple pieces and loading them instead of having them all in one script reduced the RAM usage by around 10k.  To give real-word numbers, by only refactoring, RAM usage went from 49208 to 39512 when I moved everything back into one script load per loop to get around the slowness issue in 2.2.2.

When I was testing splitting the script into multiple loads, I didn't see much of a slow-down at all on 2.2.1.  Even through 4 scripts were being loaded in each loop.  It wasn't till a user tried it with 2.2.2 that the issue was discovered.

Caching functions at init() use memory.  Memory is the limitation.  If I would use this method, the script on it's own probably wouldn't run.  And trying to do anything else would be impossible.

I don't follow what you mean by I blamed OTX without any direct evidence?  I run the exact script on 2.2.1 and it works fine.  Without doing any code changes at all, running the same script but with 2.2.2, it's **SUPER** slow.  Not just the script but the entire UI.  Maybe that wasn't communicated correctly?  If I change NOTHING but the version of OTX and it becomes unusable, what else could the problem be other than OTX?  My instructions for duplicating have you try the same script with both 2.2.2 and 2.2.1 and notice the difference in speed.  If that's not 100% evidence of the problem being in OTX, I don't know what it would take for you to accept that it's an issue with OTX.  I really don't think you understood my original post as I give 100% evidence the problem is OTX.

teckel12, 2018-08-26:
@mpaperno I appreciate your work with the `loadScript()` feature.  Trying to write a lua script with lots of required trigonometry functions and a full-featured UI is quite the task.  Trying to fit it all on a small microcontroller with limited memory is also quite challenging.

Good advice on doing any type of file access inside a run/background script.  But with memory being exhausted there's little option but to do so (at the same time limit this option as much as possible).  I'll heed this warning.

Breaking things up into multiple files (in the right way) does save RAM, **HUGE** amounts of RAM.  I was down to under 10k free running my script alone.  By breaking the script up, I increased the available RAM to almost 50k free.  Enough so that both my script and a limited version fork of the Betaflight Lua script (only VTx control) would run at the same time.

I'm doing the `loadScript()` with the "T" flag only as a result of trying different options and verifying it used the least amount of memory.  As "T" is the default anyway for the companion, I left it that way figuring it didn't make a difference on the transmitter.  So I guess it was just an oversight to not totally remove it.

Don't quite follow you about the plain-text version not being in the same folder.  For development, the plain-text version very much is in the same folder.  This is how the companion compiles the lua files.  My local development doesn't look much like how it looks on GitHub, so that could be a point of confusion.  Also not sure what you mean by setting a local/global var for debug purposes.  I'm not aware of a method of doing a debug mode in the companion.  Maybe because I'm using the companion to do all the pre-compiling to the .luac for the transmitter as well?

Again, thanks for your assistance.

teckel12, 2018-08-27:
As OpenTX is bug free, no reason to leave this open.

closed 2018-08-27 by teckel12

schwabe, 2018-08-27:
@teckel12 I think you are overreacting. As Max has explained the issue you are seeing can be very real. But also we need help tracking down what **exactly** is problem. Someone (either us or you) need to narrow down the problem.

A helpful step from your side would for example be to do a git bisect and figure out which change in the OpenTX caused the 30% increase that you are seeing. Since you seem to devised a good method measuring that should be possible.

mpaperno, 2018-08-27:
[Back on topic.]

Some things to check/confirm:
* Does this happen with all/any Lua script loading (not just this one example)?
* Does it affect loading other assets besides scripts?  Graphics/sounds.
* Does it affect loading speeds unrelated of Lua? (may need firmware-level debugging)

teckel12, 2018-08-27:
@mpaperno I appreciate your assistance and I'll be looking into each of your suggestions.  I've closed this issue as it appears the issue is with the `luac` build option not 2.2.1 vs 2.2.2.  I was at the mercy of the firmware that I had saved (as I'm not aware of a method of downloading old firmware only current builds).  The 2.2.1 that I had was without `luac` checked while the new 2.2.2 firmware I was testing had `luac` checked.

In other words, this issue appears to be related to the `luac` option causing additional and seemingly unneeded IO when the script is already pre-compiled.  Due to this chance, I've opened #6130 which seems more accurate.

mpaperno, 2018-08-27:
Next time please close the issue by stating the actual reason instead of leaving what appeared to be a caustic comment.  I had written more to try to help you out but deleted it after seeing that.  And take to heart that what may seem like unshakable proof one day, may turn into a red herring after a night's sleep or a week on the shelf (been there, done that... a lot).

teckel12, 2018-08-27:
@mpaperno At the time I closed this issue I got the vibe that there was unwillingness to consider the problem was with OpenTX, so the closed comment at the time was accurate from my perspective.  But, after further posts which caused me to dig into it more, I discovered the difference was with the `luac` build switch.  I have since updated the closed reason and pointed to the new issue.

Thanks again.

teckel12, 2018-08-27:
@mpaperno 

> about the plain-text version not being in the same folder
> 
> loadScript() @param mode:
> 
> T (default on simulator) prefer text but load binary if that is the only version available.
> bt (default on radio) either binary or text, whichever is newer (binary preferred when timestamps are equal).
> So with "T" it is always loading the .lua file if it is available (that is, exists in the folder).

I understand what the `loadScript()` options are for.  What I don't think you're seeing is that in my build environment in the companion the plain-text versions `.lua` are in the same folders as the compiled `.luac` files.  The "bT" option is specified as I always want the companion to read the `.lua` files and create the compiled `.luac` files (which again are in the same folder). But on the transmitter I only want pre-compiled files to be used.

For distribution to the transmitter, file links and `.gitignore` are used to only have the pre-compiled `.luac` files in the `dist` folder.  So What you see as the plain-text and pre-compiled files not being in the same folder is not the case.  In the companion they are all in the same folder.  For distribution to the transmitter I don't do that as I never want the source files to be on the transmitter as there's no possible way the transmitter could compile the script (will run out of memory).

I hope I'm interpreting what you were trying to say and that you understand what the build environment really is and why it's done this way.

mpaperno, 2018-08-27:
So like I said originally, you're relying on the fact that someone didn't put the .lua files in those same folders on their SD card.  Which they may not even realize if they did a sync.  And then they don't tell you they did that while you're trying to help them debug why the scripts aren't loading correctly.  :roll_eyes: 

> so the closed comment at the time was accurate from my perspective.

Great, you waste our time, talk trash about us, throw a tantrum when the response isn't quick enough for you, then when it turns out to be your own mistake there's not even a "sorry."  :+1:

teckel12, 2018-08-27:
@mpaperno Huh?  Trash talk?  Throw a tantrum?  Response not quick enough?  Where are you getting this?

Please show me where I made even a suggestion that I didn't think a response wasn't quick enough.  I feel quite the opposite actually, I believe you responded quite quickly.  You can't just put words in my mouth and expect it to be true.  If there was any trash talking it has been on the OpenTX team end.

And it turns out to be my own mistake?  Well, the mistake was still OpenTX, not my own, so nice try trying to put those words in my mouth as well.  The only mistake was that it wasn't 2.2.1 vs 2.2.2, but which build options were selected in those two builds.  I acknowledged this, stated it, and opened a new issue that was more accurate.

You people are truly unbelievable.  I'm trying to help you with your project and this is the kind of reception I get?  You really need to work on your support triage skills, or take a step back from your keyboard and take a few deep breaths.  This project has become too personal to you (and others) and you can no longer judge reality correctly.  Honestly, good luck finding people willing to contribute with your poor attitudes.