opentx/opentx #6130: Seems OpenTX is doing more IO than needed with `luac` switch

teckel12, 2018-08-27:
**Describe the bug**
Loading pre-compiled lua script with `loadScript(FILE_PATH .. "config.luac", "b")` and getting the following debug output:

```
convertToSimuPath(): /SCRIPTS/TELEMETRY/iNav/config.luac -> C:/Dev/SD Card/SCRIPTS/TELEMETRY/iNav/config.luac
findTrueFileName(C:/Dev/SD Card/SCRIPTS/TELEMETRY/iNav/config.luac)
	found in map: C:/Dev/SD Card/SCRIPTS/TELEMETRY/iNav/config.luac
f_stat(C:/Dev/SD Card/SCRIPTS/TELEMETRY/iNav/config.luac) = OK
convertToSimuPath(): /SCRIPTS/TELEMETRY/iNav/config.lua -> C:/Dev/SD Card/SCRIPTS/TELEMETRY/iNav/config.lua
findTrueFileName(C:/Dev/SD Card/SCRIPTS/TELEMETRY/iNav/config.lua)
	found in map: C:/Dev/SD Card/SCRIPTS/TELEMETRY/iNav/config.lua
f_stat(C:/Dev/SD Card/SCRIPTS/TELEMETRY/iNav/config.lua) = OK
luaLoadScriptFileToState(/SCRIPTS/TELEMETRY/iNav/config.luac, b): loading /SCRIPTS/TELEMETRY/iNav/config.luac
convertToSimuPath(): /SCRIPTS/TELEMETRY/iNav/config.luac -> C:/Dev/SD Card/SCRIPTS/TELEMETRY/iNav/config.luac
findTrueFileName(C:/Dev/SD Card/SCRIPTS/TELEMETRY/iNav/config.luac)
	found in map: C:/Dev/SD Card/SCRIPTS/TELEMETRY/iNav/config.luac
f_open(C:/Dev/SD Card/SCRIPTS/TELEMETRY/iNav/config.luac, 1) = 06AC3E28 (FIL 0E6BF1C4)
f_close(06AC3E28) (FIL:0E6BF1C4)
```

**Expected behavior**
As the "b" specifies binary only and the file name is correct, I would expect only the following to happen:
```
f_stat(C:/Dev/SD Card/SCRIPTS/TELEMETRY/iNav/config.luac) = OK
f_open(C:/Dev/SD Card/SCRIPTS/TELEMETRY/iNav/config.luac, 1) = 06AC3E28 (FIL 0B65EDE4)
f_close(06AC3E28) (FIL:0B65EDE4)
```
Doing all this other IO seems unneeded and would seem to cause a huge slowdown which isn't at all necessary.  Maybe the `convertToSimuPath()` and `findTrueFileName()` is required and doesn't cause file IO, but it would seem that `f_stat()` is IO and the extra IO wouldn't be required.

It seems that when the `luac` build option is included in the firmware, additional IO is being done even when all the scripts are compiled, there's no source code files, and the `loadScript` "b" option is selected.  This results in a 30% slow-down when the firmware includes the `luac` build option compared to when the `luac` build option is not included.

The 30% slow-down is calculated by measuring fps.  With the `luac` build option unchecked, 14.9 fps was recorded.  When firmware was used with the `luac` option was checked, only 11.5 fps was recorded.

Compiling on the transmitter is usually a futile exercise as the limited memory can many times cause it to not be able to compile at all.  So, the `luac` option is of little to no use in my experience (instead, compiling with the companion or Travis are better options).  But, it also seems there's some significant overhead from the `luac` option which appears to be caused by additional and unneeded IO activities.

teckel12, 2018-08-27:
**To duplicate**

1. Copy the contents of [test.zip](https://github.com/opentx/opentx/files/2323916/test.zip) to the transmitter's `SCRIPTS\TELEMETRY` folder and run the script
1. Flash firmware with `luac` disabled and notice the fps at the top center of the script in the title
1. Flash the firmware with the `luac` checked and notice the change in fps

schwabe, 2018-08-27:
@teckel12 would you be able with that do a git bisect to narrow down to figure out which change caused the problem?

kilrah, 2018-08-27:
It is possible to specify what's desired in the case of a luaLoadScript though.
But even if you can specify "binary only" it is only taken into account after we have checked fro presence of both. 
So yes if luac is enabled on every load even of a binary script there is still a file IO operation to check for the text version.

3djc, 2018-08-27:
Just for clarification, as there seems to be some misunderstanding out there. LUAC option does not add a compiler, LUA is always compiled JIT if this option is not declared.

With this option, it gives you the option of compiling externaly and use the compiled file to prevent local compilation. But it needs to check whenever your local lua file is not more recent than the luac file.

If you have small LUA files, it might be faster to JIT compile than do SD access to check the lua file template (so you can prevent that be not using LUAC option), or you can keepe the option and remove your lua files, (but then we need to see how fast/slow the siùple file presence check will take)

schwabe, 2018-08-27:
so your script is using loadscript on every run call at least once, so you are doing 30+ i/o operations per second. Even if the two scripts are too big to be in memory at the same a good micro optimisation would to have something like lastscripttype=1 and have something like lastscript = loadscript and set that to nil and call loadscript only if your scripttype that you need changes.

teckel12, 2018-08-27:
@schwabe But with the "b" option it should only use the compiled version, so if `luac` is activated or not should be irrelevant as the scripts are already compiled.

@kilrah That would seem more logical.  If "b" (binary only) is selected, I can't think of a reason why it would look for the possibly non-existent and absolutely unneeded uncompiled file.

@mpaperno I appreciate you looking into if the `f_stat()` (and possibly other functions, like looking for a `.lua` file at all) when "b" is specified.  It would be nice if I could use `loadfile()` but the issue with doing that is that I'm using the companion to do the compile (as the script can't compile on the transmitter).  If there was a method of using different code when running on the companion and on the transmitter, I could then do a `loadfile()` on the transmitter and a `loadScript()` in companion.  Something like in C:

```
#ifdef COMPANION
loadfile()
#else
loadScript()
#endif
```

What I figured is that using the "bT" option with `loadScript()` would in essence do exactly that.  Always do a compile in companion and always use the binary on the transmitter.

I can't control what build options people selected for OpenTX.  I will be suggesting to disable the `luac` build option due to this performance hit.  But, that's not always the best idea as the `luac` option can be really helpful for smaller scripts.

@schwabe Yes, the script is loading a file on every loop.  It's doing 11.5-15.0 i/o operations per second to be more accurate.  This is not desired, but required to limit the memory used on the transmitter.  I understand it's not ideal, and should be avoided, which I have done till the latest release of my script.  But, The memory reduction benefits can be quite incredible, going from less than 10k free to almost 50k free.  My testing was with the `luac` switch off, so the overhead wasn't very evident.  But with the `luac` switch on there's a large performance hit (even when the code is 100% pre-compiled).

I like your idea on noting the `lastscripttype` and only doing a loadscript if it changes.  I've actually started looking into something similar last night.  As I said, this performance hit wasn't so evident till I used the `luac` build option, so with the `luac` build option turned off there wasn't an issue to address.  But yes, now knowing the huge performance hit with the unused `luac` option, an alternatives are being looked at.

schwabe, 2018-08-27:
I/O is slow. I/O is always slow. That is one of the first lesson people should learn. Modern OS are doing all kind of tricks and using a lot of memory and CPU to cache and avoid real I/Os, so lazy programming and doing more I/O than needed works with modern OS. If your OS (windows or mac) would not cache the files from disc in memory you also would see a very slow loadscript on them.

On the small Taranis I/O is even more expensive and the small CPU has no memory to spare to do any caching, that is why this lazy programming really hits you hard here.

Yes we can optimise loadScript a bit but that only masks the real problem here.

schwabe, 2018-08-27:
@teckel12 The 50k free might be a red herring. As you will only see memory between iterations you might actually use a lot more memory while the script is running. and since you forcefully gargabe collect the loaded script, the memory is free'd.

kilrah, 2018-08-27:
To be clear you also see a "performance hit" whithout the luac option. Reloading the same script you were already using does take the 20ms or so it takes to read from SD and saves no memory. Aka once you balance your thing a bit more sensibly it'll likely be way faster.

>If there was a method of using different code when running on the companion and on the transmitter, I could then do a loadfile() on the transmitter and a loadScript() in companion. Something like in C:

There is, and @mpaperno gave it to you in the other issue...

> Why? Seems running >10Hz is kinda pointless in the first place.

teckel12, 2018-08-27:
@schwabe Let's put this in perspective.  I've been writing the Lua Telemetry script for about a year now.  Up until the last release, I wasn't using the `loadScript()` function at all as being a programmer for 37 years I'm well aware of the performance hit of I/O, and have a lot of experience with microcontrollers of limited speed.  The `loadScript()` was introduced just in the last week to try and reduce the memory used to a point where both Lua Telemetry and a fork of the Betaflight Lua Script could run on the same model at the same time.  I was doing my testing with the `luac` build option turned off, which didn't uncover NEARLY as much of a performance hit.  A user had the `luac` option turned on, which resulted in an EXTREMELY laggy UI.  As there's more I/O, the slowdown with the `luac` option turned on is compounded.

So, I would suggest that the last year of programming Lua Telemetry exclusively and the 609 commits is probably a good indication that the issue here is not lazy programming.  The issue is that there's a huge performance hit when the `luac` option is enabled.  Sometimes, programmers need to do things that are maybe not the best but are required as the only option.  The reduction of used memory is something that I've been working on for almost a year.

I appreciate you attempting to school this new programmer with your 101 class, but this is only causing the focus of this issue to be misdirected as my skill as a programmer and experience with microcontrollers and I/O is not the issue.  The issue, if we could focus on it, is the 30% slowdown when using the `luac` build option.

teckel12, 2018-08-27:
@kilrah Sure, there's a performance hit when adding any additional code, no matter how small.  But that isn't in question or the issue here.  The issue here is that the exact same code is 30% slower with the `luac` option enabled than when disabled when I don't believe there should be as the code is all pre-compiled binaries.

I didn't get mpaperno's code suggestion about companion detection till after I posted it here.  Maybe obvious, but wanted you to know I'm not asking the same thing multiple times, I simply didn't see his post in another issue thread till after posting what I did here.

I will be suggesting to disable the `luac` option because the 30% fps slowdown is just something easy to measure.  There's other issues that users are reporting as well, like voice playback is choppy when `luac` is enabled.  In other words, I'm showing just the fps thing as that's easy to reproduce and show that it's really happening.  But, there's other issues that are resolved with `luac` disabled because it's more than just a fps hit.  I hope this makes sense.

schwabe, 2018-08-27:
Lazy programming was meant as programming paradigm, you probably have heard about that, for example in the context of lazy evaluation. I was trying to bring my point across.

@teckel12 as @mpaperno said the loadfile command exists which you can use.

Other than that, you just very lucky that loadscript only uses < 30 ms without luac.

You are asking us to optimise for your corner case of which nobody of us think that is a good idea to do anyway and you have been given an alternative.

For running the scripts in parallel. It might be good a idea to check if you can make both screen to only load the main scripts if they are actually being displayed. Loading/unloading one (or even both) script every time it is run is problematic as you have seen.

I understand that fixing it the right way will be more challenging than just blaming it on us and that luac takes 30% longer. But also 30% longer loadtimes in loadscript also means that you are already at a point where small changes (like this 30%) or a slower SD card (the one that came with my X9D+ can only do 600kB/s) can also screw up your lua script.

kilrah, 2018-08-27:
My first post in this issue points at and ackowledges the likely reason for the performance drop. We'll  improve that at some point when we have time. Not saying "issue" or "fix" since loading a script many times per second has never been an envisioned use case and it has never been done by anybody else.

The rest of the discussion was trying to make Mr 37-years-of-experience know-it-all understand that regardless of that issue that there are still significant and rather obvious improvements you can do on your side to tax the system less at which time your problem would likely disappear and that are good to do regardless of whether OpenTX behavior is changed or not, but it seems that point has less chance of succeeding.

Note that with your experience, and the source of the performance drop being pointed to you you should be perfectly capable of making the change in OpenTX yourself and supplying us a pull request - at which time your problem would disappear in next OpenTX version instead of having to wait un indeterminate amount of time until we have time to take care of it.

teckel12, 2018-08-27:
@schwabe 
> voice playback is choppy since you cannibalise almost all of SDcard I/o for your script, so the sound play does not get enough i/o to play sound without running out of data.

But keep in mind, this doesn't happen with the `luac` option enabled, hence the reason for this issue.

Also, even when there's not a `loadScript()` in a loop, simply doing a few `loadScript()` commands at init (which is is a common practice with lua scripts) causes the robotic voice if the voice is triggered at the same time as init (and again, only when the `luac` option is used even and even if the script is already compiled).  An example of this is I have a voice file for each of my model names.  When I select a model, it reports the model name.  When there's a script running that does a few file loads at init (and the `luac` option is checked) there's robotic voice for the model name, even when the script is already compiled.  If the `luac` build option is turned off and the same is repeated, the model name plays clearly.

So it's not a problem ONLY when there's a `loadScript()` inside a loop that's hammering away creating all kinds of I/O access.  `luac` also creates lag when doing a few `loadScript()` at init, and so much so that it causes robotic speech.  I would assume doing `loadScript()`s at init would not be considered "lazy programming" to you?  If so, I can create another issue with different example code that will show you the problem with the `luac` option even when unlazy programming is implemented.  I'd suggest that isn't needed as this current issue (if you ignore the noise) is quite clear already.  But, if it isn't obvious already, I'm not unwilling to invest time into this.

kilrah, 2018-08-27:
You really have a hard time reading. We said we understood the issue and will consider a related improvement. This github issue is perfect for tracking this.

We're also telling you you can do better with your script and even telling you how. You're loading tiny scripts like reset.luac, modes.luac and load.luac. I acn't see how doing it with these tiny ones would have any meaningful memory saving, BUT each time you load one of these you're putting a ~30ms delay in your execution.
So yes loading scripts is a good idea, BUT only on heavy ones, then avoid loading those you don't need, or unload something to reload it again just after.
**Even if your fps is higher without luac you are also currently putting too much load on the system.**

Really don't know how to put it more simply for you to understand all you've written here was pointless becasue we HAVE understood and acknowledged your problem long ago already.

schwabe, 2018-08-27:
@teckel12 Let me give you an advise. Let someone you trust and is somewhat neutral (like someone from the inav team?) read this discussion and let him/her give you/us feedback what they think. This discussion here like @kilrah says isn't anyway since we are not changing our opinion and we also cannot convince you.

teckel12, 2018-08-27:
@kilrah I appreciate your coding suggestions and as I've stated I've already implemented some of them and working on the others.  Keep in mind, this refactoring was not required with the `luac` build option enabled.  Please see my quote above:

>I'm not thrown my hands up in the air and will not change another line of code till something is resolved with OpenTX. The truth is quite the contrary. I've already created a new release version 1.4.1 which removes most of the i/o and I've created a `refactor` branch to deal with the suggestions posted here.

I've been dealing with the limited remaining memory on the transmitter for a year now.  And loading those tiny scripts like `reset.luac`, `modes.luac` and `load.luac` make a BIG difference once added up.  I went from under 10k free to almost 50k free by piecing things up like this as the garbage collection can happen between.

A little history, at first I had it load one script on startup, but that didn't reduce the memory at all (it actually made it use more memory).  I then tried to break the load up into two scripts and that's where I saw a memory used improvement.  So I split it up further and recorded more memory improvement.  By creating several loads where garbage collection can happen in-between, a LOT of memory can be saved.  I have pages of memory tests recording the savings in memory from one step to the next.  So I didn't just do this without testing it every step of the way.  There really is substantial memory saving using this approach, and other's who have suggested methods to save memory don't actually result in memory saved.

If you'd like, I could create two versions of the exact same code.  One as it is now with the multiple loads and another refactored to have them all load at once.  The difference is quite significant.

I'm not the only programmer taking this approach.  If you look at the Betaflight Lua Script it has **many** `loadScript()` events on init.  This is becoming standard practice with large/complicated scripts due to the limited memory available.

teckel12, 2018-08-27:
@kilrah 
> Even if your fps is higher without luac you are also currently putting too much load on the system.

Noted.   And as I've said several times now, I'm look into ways of dealing with this without using more memory.  The `refactor` branch is exactly that.

kilrah, 2018-08-27:
> I went from under 10k free to almost 50k free

We still think you're misled by something. When I load a blank model with your script assigned my "never used" mem goes down to 43K, after the script is displayed once 31. So as others mentioned whatever you're looking at is likely sampling at a time that does not represent anything meaningful.

teckel12, 2018-08-27:
@kilrah I am looking at the same numbers as you, in the same way.  I'm not looking at sampling at a time.

If I would take that same code but refactor to load as one script, it would probably be closer to 10k free using the same process as you are now.

I know this is a bit off-topic.  But I'd be more that willing to do that for you if you'd like.  Shouldn't take long.  And maybe this would allow you to understand the reason why this is required.

kilrah, 2018-08-27:
Let's jsut forget it. 

We work on avoiding unnecessary I/O with luac.
You work on avoiding unnecessary I/O in your scripts. To this purpose you may use some our suggestions like loadFile().

teckel12, 2018-08-27:
@kilrah Really appreciate the suggestions.  Using `loadFile()` on the transmitter only very may very well resolve the issue till such time 2.2.3 is released.

As I already refactored the code to run in one file I've included it here:
[memTest.zip](https://github.com/opentx/opentx/files/2325248/memTest.zip)

I only ran it in the companion as I don't currently have a transmitter with me.  It runs fine in the companion, but there very well could be not enough memory to run it at all on the transmitter.

I'd be interested to see what you find as the memory used as what I have experienced is a HUGE memory reduction.

teckel12, 2018-08-27:
@kilrah You experienced 43K free at startup and after the script is displayed once 31k free.  Can't wait to hear back what happens when not using `loadScript()` to load all of those small files!

teckel12, 2018-08-28:
@kilrah So I'm with a transmitter again and I ran the two sets of code who's only difference was loading as multiple scripts or running as one script.

With multiple scripts, on startup the free memory was 39,572 and once the script was displayed the free memory was 30,712 (fairly similar to your results).  I then ran the memTest script which is exactly the same code, just refactored into one script.  The free memory on both startup as well as once the script was displayed was 7,248 free memory.

This, however, is with a reduction in the number of scripts as I had already removed many which increased the free memory another 10k/7k.  But still, it shows that with skilled programming, you can indeed save a LARGE amount of memory by breaking a script into smaller pieces.

The totals (even with this crippled example that doesn't even show the total potential) is 32,324 saved on startup and 23,464 saved once the script was shown.  And remember, these numbers were close to 10k/7k higher with all of my enhancements.

I realize it can be humbling to realize you don't know everything, but it takes a big person to admit when they're wrong and accept that in some situations others are the experts.

Anyway, it seems both projects have grown from this.  It didn't need to happen the way it did, but I've taken much of your advice to heart and made numerous changes and it seems OpenTX has also been enhanced as a result.  There's a much easier way to approach this in the future, and I hope you could see  this as well.

closed 2019-11-15 by stale[bot]