opentx/opentx #6132: Crossfire Telemetry - Sensor lost alarm in 50Hz mode

lostcontrol, 2018-08-28:
I switched to Crossfire a few days ago on my Taranis X9D+ and noticed that it often reports "sensor lost". Looking at the problem, I noticed that when Crossfire switches to normal mode (aka 50 Hz mode), it sends telemetry data for "external" sensors (like Betaflight ones in my case) at a much slower rate like every 15-20 seconds. In the Crossfire manual, it is stated that in 50 Hz (normal mode), "It has also telemetry support in both directions, but only for the most important information i.e. GPS, current sensor, attitude, and simple commands." So it seems here that "only for" means at a much slower refresh rate.

TELEMETRY_VALUE_OLD_THRESHOLD is set to ~10 seconds and therefore the Betaflight sensors are repeatedly reported as lost when Crossfire runs at 50 Hz.

You can disable telemetry alarms in OpenTX but then you get an RSSI warning screen every time you turn on your transmitter or change to that model. For now, I put an empty sound file for sensorko.wav but that's a rather bad workaround.

There is already a few bug reports about too slow refresh rate of the telemetry sensors but I think for Crossfire, it is a design decision and something that should be taken into account.

**Expected behavior**
It would be nice to have the timeout adapting automatically to the Crossfire mode (150Hz vs 50Hz) but that's probably a bit complicated.
It would help to be able to set a specific timeout per sensor or to suppress the warning per sensor (I care if I loose e.g. LQ sensor but not really if I loose the current sensor).

**General information**
 - OpenTX 2.2.2
 - Frsky X9D+
 - Crossfire Micro TX/Nano RX

kilrah, 2018-08-28:
If the CF manual says only the "most important information" is sent then I would understand that something else being sent once every 15-20sec is a bug on their side (it's useless to have data that slow anyway).

I'd think the correct fix would rather be to have a switch in betaflight to not send those sensors when in 50Hz mode.
On your side you can simply delete those sensors on the radio and they won't be taken into account.

lostcontrol, 2018-08-28:
I'm new to Crossfire so I don't really know if this was always like this or not. I just updated my Micro TX to the latest 2.40 which came out a few days ago. Friends flying KISS just told me they never had the "sensor lost" alarms so I'm wondering if this could be something in Betaflight too!? I looked in the code there but it seems there is no difference if 150Hz or 50Hz. Seems like telemetry on ßF side is sent 10 times per second. Could also be the 2.40 XF firmware since it is pretty new and I guess none of my friends flew it yet.

Yes, of course I could delete those sensors but the problem is I want them (at least the battery voltage). If I fly LoS, I like to have that on my Taranis. And flying LoS, I'm pretty sure to be in 150Hz mode so the sensors are fine.

Is it uncommon to have sensors refreshed at different rates?

kilrah, 2018-08-28:
Different rates no, but not that slow. It makes no sense at all to send as rarely as 15-20sec, data is useless when it's that old and savings are negligible. You might have stuff you send at 10Hz and some only once a second, but not less frequently than that.

Someone else has reported a problem with crossfire telemetry since they updated to the latest betaflight, so there might be a bug in BF.

lostcontrol, 2018-08-30:
Here is a video showing my problem: https://www.youtube.com/watch?v=9pUMfzDf0_8

Again, I don't think the actually slow refresh rate as anything to do with OpenTX. Just that some more flexibility in the sensor configuration would make it possible to workaround that problem.

lostcontrol, 2018-08-31:
Yes, but then I get that "RSSI warnings" screen every time I turn on my Taranis. For now, I emptied the "sensorko.wav" file but I realized yesterday evening that the haptic vibration also gets triggered. I might go for the disable all telemetry alarms options in the end...

I have the LQ monitoring setup in my Taranis already.

I got some feedback from people that also get this. Might be related to Betaflight also. If you don't mind, please, keep this ticket open for now and I'll close it as soon as things are clarified. Thanks.

3djc, 2018-08-31:
As I said, you can disable telem alarm altogheter, and make your specific alarms for things that matter to you, while either BF or CRSF folks fix the issue

sirPerna, 2018-08-31:
I don't think the slow update rate is an OpenTX issue. We will have a look at this from the Crossfire side.

fixmania, 2018-10-02:
Is this solved? I have a same issue. I think Betaflight 3.5 problem.

lostcontrol, 2018-10-03:
@fixmania no, I still have the issue. I opened a ticket at TBS but they keep telling me that "in 50Hz mode, telemetry throughput is decreased, some sensors have less priority, etc, etc, etc". They never answer my questions regarding if they tried to reproduce the issue or if they got other reports. As you say, could also be a problem with Betaflight. I didn't try to downgrade to 3.4 to see if it is also happening there.

lostcontrol, 2018-10-03:
The best you can do guys is probably to open a support ticket on TBS support website and tell them about your problem. Maybe they will then investigate the issue when several people start to complain about it... Feel free to mention the ticket I created over a month ago there:

`#741290 Crossfire / Telemetry update rates in 50Hz mode`

sirPerna, 2018-10-09:
@lostcontrol can you tell me which sensor get lost? OpenTX indicates the lost sensor with [ ] when you go to the telemetry menu page.

lostcontrol, 2018-10-09:
@sirPerna sure! The Betaflight sensors get lost, happened with all of them (at least the ones I'm using). That's why this could be also a ßF issue, since I have quite a few friends with XF and KISS that never got this issue (not sure they are all using KISS telemetry e.g. battery voltage on their Taranis though).

You'll see some of my ßF sensors getting not refresh in time in this video: https://www.youtube.com/watch?v=9pUMfzDf0_8

javaboyjunior, 2018-12-20:
I have attached a csv log file if useful. My Taranis was buzzing approx every 10s (have already disabled the wav file) and I can't figure out which sensor is getting lost even with the csv. I would remove that sensor if I could figure out which one is the issue.

Suggestions much welcomed. Thanks!

[Dronez-2018-12-20.csv.zip](https://github.com/opentx/opentx/files/2700421/Dronez-2018-12-20.csv.zip)

sirPerna, 2019-02-25:
This issue is fixed by V2.93. It will be available for public beta within a few days.

3djc, 2019-02-25:
Thank you for the update, closing this then as it is not an OpenTx issue

closed 2019-02-25 by 3djc