opentx/opentx #6134: Feature Request: Generate event trigger after each cycle through telemetry sensors

teckel12, 2018-08-28:
**Is your feature request related to a problem? Please describe.**
Telemetry lua script screen updates much faster than telemetry is being received, therefore using microcontroller resources without benefit.  Instead, allow the ability for the UI of a telemetry script to sync to the telemetry received refresh rate.

**Describe the solution you'd like**
A lua exposed event that would trigger when OpenTX receives one "cycle" of telemetry.  Basically, it's kind of pointless to paint the UI at 20 fps (the default) when telemetry is only being received at maybe 2 - 5 fps.  Much like a button event, this telemetry event would allow a lua script to only update the screen when there's new data to display.  This would return resources to core OpenTX functions.

**Describe alternatives you've considered**
Creating a new telemetry sensor from the flight controller that would increment.  This counter would be used as the trigger to initiate a screen update.  While this would work, it would need to be added to each flight controller and adding a sensor would slow down telemetry.

**Additional context**
Haven't looked at the OpenTx code in how it receives telemetry sensor data and how the current key event structure could be used.  Throwing this feature request out there for those familiar with the OpenTx code to see if this is feasible or if you believe it would actually be useful.

I believe the way (at least with S.Port) telemetry sensor data is received is that it doesn't go from sensor 1 to 20 or whatever.  Instead, the flight controller [can] send some sensor data more frequently and others less frequently.  So, there could be no real order to measure when a cycle is complete.  If this is the case, I believe the way to denote a cycle is when OpenTx receives RSSI telemetry.  But, maybe this is a naive guess.

kilrah, 2018-08-28:
There are no such things as "telemtry cycles" as explainesd in your other issue already.
You should check for value change in your scripts.

closed 2018-08-28 by kilrah

kilrah, 2018-08-28:
No, forcing a base on an arbitrary sensor that has no correlation whatsoever to the others is not a good way at all.

teckel12, 2018-08-28:
@kilrah Sorry, thought it could have been an elegant way to throttle resources.  I'll look at writing the code and submit a PR.

kilrah, 2018-08-28:
You are the one writing the script, you are the one who knows what data is important enough to trigger an update for your script. When you get that set a flag, if flag is not set just return without doing anything when it's time to draw.

teckel12, 2018-08-28:
@kilrah Well, all the data is important of course.  As a telemetry script is tied to telemetry data being updated, it seems logical to sync display updates to when telemetry data is updated.  For example, with my script, there's close to 20 telemetry sensors that are all important to be looked at and the display updated when changed.

And I wasn't at all demanding that someone write this for me.  If that's what you thought you're mistaken.  It's just an issue, something that can be discussed and a possible solution found.  I had fully planned on writing the code myself.  But getting feedback via an issue typically makes the code better via collaboration.

kilrah, 2018-08-28:
> As a telemetry script is tied to telemetry data being updated, it seems logical to sync display updates to when telemetry data is updated. 

Then syncing on RSSI is not the right solution in the first place... you're just saying you need to sync on 20 values, so you would end up redrawing as often as all 20 will come at different times.

So run your script at 5Hz and call it a day. 

...or go see what others do and how someone like C&T can run a complex script at 20+Hz with actual telemetry updates without any resource issues.

teckel12, 2018-08-28:
@kilrah I believe RSSI comes in the most frequently, which is why it's the prefect trigger.

I've tried running it at 5 fps, but that can almost guarantee lag while having it synced with speed at which telemetry is received would reduce that.

I would argue my script is one of the most complex, and it runs at 20 fps as well.  It's so complex, that I'm looking for creative ways to fit it all in the memory available.

kilrah, 2018-08-28:
You believe wrong, RSSI comes at 4-5Hz. Others can come at 10+.

schwabe, 2018-08-28:
@teckel12 I gave a very short look at your script. You can improve memory usage a lot if you don't store all telemetry to a global variable. Calling getValue whenever you need is perfectly fine.

teckel12, 2018-08-28:
@kilrah Do you know which can come in at 10+?  It seems that the raw AccX, AccY & AccZ values come in quite frequently.  Well, they did till I removed them from INAV.

This is how it's supposed to work, an idea is raised, and a discussion can conclude it's a good idea, not focused correctly, or maybe just a terrible idea.

teckel12, 2018-08-28:
@schwabe I don't believe I'm using any global variables (all are set a local).  But maybe you mean variables that are used throughout the script as "global"?  I'm guessing you mean something like this:

```
data.altitude = getValue(data.altitude_id)
```

And instead I should just use `getValue(data.altitude_id)` instead of storing the altitude in `data.altitude`? Seems logical that it may use less memory that way, but what about performance?  What's the CPU resource hit by calling getValue() a bunch of times instead of a variable?

BTW, thank you very much for the suggestion!  Feel free to suggest anything, or even create an issue.  I love this kind of feedback!

schwabe, 2018-08-28:
getValue is written in C and is probably around the same than using data.altitude or might even be faster since a lua dict lookup is also not super fast

teckel12, 2018-08-29:
@schwabe Sorry, I tried it and it actually used a bit more memory 😞 

https://github.com/iNavFlight/LuaTelemetry/issues/172

Please don't be discouraged by this, I really thought (and hoped) it would work.  If you have other suggestions or criticisms, I'm serious that I'd love to hear them.