opentx/opentx #6135: Feature Request: Generate event trigger each time RSSI telemetry is received

teckel12, 2018-08-28:
**Is your feature request related to a problem? Please describe.**
Telemetry lua script screen updates much faster than telemetry is being received, therefore using microcontroller resources without benefit.  Instead, allow the ability for the UI of a telemetry script to sync to the telemetry received refresh rate.

**Describe the solution you'd like**
A lua exposed event that would trigger when OpenTX receives new RSSI telemetry.  Basically, it's kind of pointless to paint the UI at 20 fps (the default) when telemetry is only being received at maybe 2 - 5 fps.  Much like a button event, this telemetry event would allow a lua script to only update the screen when there's new data to display.  This would return resources to core OpenTX functions.

**Describe alternatives you've considered**
Creating a new telemetry sensor from the flight controller that would increment.  This counter would be used as the trigger to initiate a screen update.  While this would work, it would need to be added to each flight controller and adding a sensor would slow down telemetry.

**Additional context**
Haven't looked at the OpenTx code in how it receives telemetry sensor data and how the current key event structure could be used.  Throwing this feature request out there for those familiar with the OpenTx code to see if this is feasible or if you believe it would actually be useful.

kilrah, 2018-08-28:
Stop spamming stuff NOW. Read the answers you're given. Let them be discussed before going further.

closed 2018-08-28 by kilrah

teckel12, 2018-08-28:
Dick

teckel12, 2018-08-28:
@kilrah Sorry, just created a different issue to be more correct based on your comment.  Didn't mean to work you into a froth.

schwabe, 2018-08-28:
> Dick

You just disqualified yourself