opentx/opentx #6137: Invalid or empty Bluetooth data stream on 2.2.x

MarenB, 2018-08-30:
Hi guys,

mentioned it already here #3852, but that actual topic seems to be solved, meanwhile - BT is advertised in 2.2.2.

My BT master (automatic antenna tracker for FPV) still connects to te radio after uprading to 2.2.x, but it receives invalid or empty data. I first noticed with 2.2.0 RC3 (BT enabled via Companion, since it was not advertised on the radio at that version), but the result is same with latest release 2.2.2.

If I switch back to 2.1.9, the data is sent out correctly and my AAT can get the model's GPS position within seconds.

**Expected behavior**
The bluetooth data stream is expected to consist of mirrored S-Port data as it was with older versions of OTX.

**General information**
 - tested with OpenTX version 2.2.2 and 2.2.1 RC3 (NOK) and 2.1.9 and older (OK)
 - Transmitter: X9E, Receiver: various, but i.e. X6R
 - BLE receiver module HM-11 as BT master

3djc, 2018-08-30:
It was probably broken during the big BT rework for the new BT module on Horus/x7/xlite. This specific issue is going to be chalenging for us to fix beacause of lack of testing equipement, but we will try our best (we may need a lot more tesitng help than usual)

marcels71, 2019-08-24:
Just a question regarding Bluetooth module in X9E. Will again work in 2.3 like in 2.1.9? Now is BT module in X9E for nothing. Thanks for comment

kilrah, 2019-08-24:
Were you doing something with it in 2.1?

The module in the X9E basically IS useless since it isn't compatible with anything else in the FrSky lineup.

closed 2020-07-08 by stale[bot]

MarenB, 2020-08-26:
> Were you doing something with it in 2.1?
> 
> The module in the X9E basically IS useless since it isn't compatible with anything else in the FrSky lineup.

I can't agree, it was very useful to stream S.Port-data to other (non-FrSky-) devices. Of course, not many people did actually do that, but I still only have the choice to stick on old firmware or install a 2nd, independent BT-module to run my antenna tracker.