opentx/opentx #6138: Blinking LCD Display in LUA scripts on Horus X10, OpenTX 2.2.2

azaz44, 2018-08-30:
**Describe the bug**
The documentation says, in OpenTX 2.2.2 LCD contents are preserved between run() functions. This means you could only draw things once, and then only redraw if something is changed.

This works fine in Companion simulator but does not work in Horuse X10. It causes a screen to blink every cycle, between what you have drawn, and the previus LCD state. It looks like it has two switching LCD buffers and drawing updates only one of them.

**To Reproduce**
1. You need to test it on physical radio. In companion everything is fine.
2. Just run the following one time script.
3. Expected behavior: seeing white screen.
4. Real behavior: Screen is blinking between white, and the system menu from which you started the script.

```lua
local needsRedraw = true

local function init()
end

local function run(event)

  if needsRedraw then
    needsRedraw = false
    lcd.drawFilledRectangle(0, 0, LCD_W, LCD_H, TEXT_BGCOLOR)
  end
  
  return 0
end

return { init=init, run=run }
```

**Screenshots**
No screenshots, can make a video if required.

**General information**
 - OpenTX version: 2.2.2. Happened in 2.2.2RC5 as well.
 - Hardware: FrSky Horus X10s
 - OpenTX Companion: 2.2.2 on Windows.

azaz44, 2018-08-30:
I'm just in the middle of some bigger development and would appreciate info if there's any workaround for that other than drawing everything every cycle. I could probably draw things twice, but it's complicated to code in my setup, maybe there is some better option?

kilrah, 2018-08-30:
The doc is correct for Taranis radios, but the color screen ones work very differently and indeed have dual buffers that are swapped at every refresh.
Not sure we can do much there...

azaz44, 2018-08-30:
Okay, how should scripts be developed then - draw everything always? Or is drawing for two cycles the right way (to save CPU)? I can try to implement the latter for my screens, but I don't want to step into some hacky way of doing things, which might stop working in future versions or would not work on Horus 12. So wondering which way is correct.

kilrah, 2018-08-30:
Drawing always.

azaz44, 2018-08-30:
Ok, thanks for the support.

closed 2018-08-30 by azaz44