opentx/opentx #6139: Navigate pages with rotary dial / scroll on Horus (possibly other with the wheel)

azaz44, 2018-08-30:
This is a wish.

I find current way of navigating between pages (press and release `PgUp/Dn` for next page, hold for previous page) cumbersome. Especially if having many pages (like in model settings), you either have to hold a lot or cycle to all pages all the time. I believe it could work much better with a rotary dial, like this:

- press and quick release of `PgUg/Dn` would switch to the next page, like it does now.
- press and hold of `PgUp/Dn` would do nothing unless rotary dial is used.
- using rotary dial, while `PgUp/Dn` is being hold, would switch between pages (in both direction).

This could be an additional build option, so you would have a choice.

One would also need some suppot for LUA scripts so they would behave consistent. Something like Page Up / Down events ready to use.

azaz44, 2018-08-31:
Some note. I'm new to OpenTX but I'm a software developer and could possibly help in future if time will allow. But wondering if such feature would be welcome?

kilrah, 2018-08-31:
Sounds good to me, just slightly annoying that there is a conflict with the PAGE button currently acting on press and not on release so that it does need a setting and can't be just added as additional possibility all the time...

azaz44, 2018-08-31:
I didn't get it, where would be the conflict?

Maybe I explain better - as it works now, both page next / prev work on release, but it makes page next when holding time was short, and page prev when it was long. This handling would stay. And for page-next there would be no difference. The difference would be, if you hold long enough, instead of page-prev it would enter the mode where you can scroll pages with the wheel. And it would be in this mode until you release the page button.

kilrah, 2018-08-31:
Ah yes it's on release, it's on X12 that it's on press.

I'd have wanted people to have both options without any setting but it's probably going to be difficult or confusing.

azaz44, 2018-08-31:
Without setting it's not possible I think, because it has to kill page-prev behaviour on hold. But I think a build option would be enough?

closed 2019-11-15 by stale[bot]