opentx/opentx #6140: Warining - Cannot Open joystick, joystick disabled

Cybershead, 2018-08-30:
Hi,
I could not see this on the forum, unless i missed, if i did apologies.

I am using 
Companion 2.2.2
Taranis X9E
Windows 10

Once opened the Radio Simulator and it pops up with **Cannot open joystick, joystick disabled.**
In Settings, simulator Settings, i have joystick enable box ticked, but the box to the left is saying no joystick found.

Any thoughts?

Thanks in advance.

mpaperno, 2018-08-30:
So is there a JS actually plugged into the computer?  If it can't find the JS to show in that list box, then it makes sense that it cannot open it either.  What kind of JS are you using?

Cybershead, 2018-09-01:
Sorry moaperno, I missed your reply. I am using the X9E Transmitter and connect the the Pc open TX via a USB. I can upload and download, so I know the USB fine. But if I simulate the Transmiter it comes up with that error.

mpaperno, 2018-09-01:
No worries.  But I'm not sure what you're trying to do.  To be clear, the Joystick stuff in Simulator is for controlling the simulated radio with a JS connected to the PC (vs. using the Tx as a JS in other apps).  

Assuming you already knew that and _are_ trying to use the Tx as a JS, the next question is: Does the JS (Tx) show up in your Windows devices list and can it be properly tested/calibrated from there?

Cybershead, 2018-09-01:
Maybe it’s me, I always thought that when I have the simulator winddow open and the Transmiter connected, it would simulate my movements on the transmitter.

mpaperno, 2018-09-01:
It should work if you have the Tx in Joystick mode (set up in radio settings as "ask" or "joystick") and your computer recognizes the Tx as a JS.  I'm guessing one/both of those isn't the case, since otherwise it would show up in the simulator JS settings list of detected devices. :)

Cybershead, 2018-09-01:
This is my issue, it will not recognise my TX Joysticks or switches. This all come about when I brought Real Flight 8. I needed to make sure the TX was doing everything.

mpaperno, 2018-09-01:
Then you're going to need to be more specific.  What have you tried, what are your radio settings, how are you plugging in the radio (do you get the "device found" chime in Windows?), etc.  This isn't really a support forum, and there haven't been any other valid reports of bugs with JS mode for a long time.  Only recent issues were from people just being confused about how to activate it.

Have you followed these steps?  https://github.com/opentx/opentx/issues/4815#issuecomment-359097130

Cybershead, 2018-09-01:
This used to work, but then again I was using a Mac. But i upgraded to 2.2.2 on the PC, hence I thought it may be a bug. Thanks for your response anyhow. I will sort it, if not a issue.

closed 2018-09-01 by Cybershead

kilrah, 2018-09-01:
You must configure the USB mode to joystick (or Ask then select joystick when plugging the cable) for the radio to act as a joystick in 2.2.