opentx/opentx #6143: Global variable pop-up issue with the Companion running Horus profile

Oldgibber, 2018-09-01:
It is not possible to enable the pop-up when programming global variables on the Horus transmitter itself. However, this feature is still enabled on the Companion. When enabled via the Companion it still does not work on the transmitter, so I presume this feature is not actually available for the Horus. Is it needed anyway?

If this is correct, then the pop-up option should be removed on the Companion when a Horus transmitter is selected.

kilrah, 2018-09-01:
The feature needs no sense on Horus since you can use widgets to display them at any time.

Oldgibber, 2018-09-02:
You've missed the point. If the pop-up option does not apply to the Horus, then it needs disabling in Companion.

3djc, 2018-09-02:
Which is what he just changed into companion, as referenced just above

Oldgibber, 2018-09-02:
Sorry guys!

StefanKellerAC, 2020-05-15:
in Companion 2.3.7 and X10 it is still selectable.

commit 55af1b0a (2018-09-05): [Companion] No GVAR popup on Horus (fix #6143)