opentx/opentx #6146: Enable bind Frsky R9 mini and slim with PWM channel 9, 10, 11, 12

davidngrc, 2018-09-03:
In Frsky Receiver X8R, the physical PWM pin, we can bind with PWM ch 1-8 or ch 9-16
can we do the same for R9 mini and slim or maybe the whole R9 family?

if this is possible, we can use the 4 PWM in the receiver to do some gimbal control or other control.
otherwise, SBus, FPort user will have 4 PWM pin wasted.

3djc, 2018-09-03:
This is not an end users support place. Please go to places like regroups or openrcforum. Alternatively you could just try on radio or simu ...

closed 2018-09-03 by 3djc

kilrah, 2018-09-03:
Module/receiver features are not under our control.

davidngrc, 2018-09-04:
I am new here, do you guys know how to contact frsky for this?
because I try to email their support before, no reply.

kilrah, 2018-09-04:
We have some contacts, but as mentioned in this case it is not needed since it seems what you want already exists.