opentx/opentx #6152: One German wav incorrect in SD content 0018

kreidler, 2018-09-04:
Sorry, but I am not able to rebuild the SD content. Therefore please correct 
in generated file "SOUNDS/SYSTEM/de-DE-taranis.csv" the line 180 from

SOUNDS/de/SYSTEM;sensorko.wav;Sensor veloren
to
SOUNDS/de/SYSTEM;sensorko.wav;Sensor verloren

Thanks

closed 2018-09-05 by 3djc

kreidler, 2018-09-06:
@3djc :  Downloaded nightly sdcard content dated 2018-09-05. Issue still existing.

3djc, 2018-09-06:
Yes, I think I know why (it is fixed, but that fix did not reach the server generating the sounds when I regenerated those sounds. Will post here tomorrow when redone

3djc, 2018-09-07:
@kreidler , it should be ok now, sorry for this.

kreidler, 2018-09-07:
@3djc : Thanks, it is ok now. Do not apologise that's software delevelopment

commit b4be6158 (2018-09-05): Fix german sensor lost audio
This fix #6152