opentx/opentx #6156: Setting Trainer Conditions

geekerator, 2018-09-05:
It does not seem to be possible to set:

Radio Setup > Trainer > Ail (or Ele, Thr, Rud) mode(s) to: OFF, :=, or += with a special function in 2.2.1 or 2.2.2

This is a problem for me when using a head tracker (with pan/tilt on channels 5/6) via the trainer port (which requires the "OFF" mode for each flight control) followed by using the trainer port for a buddy box (which normally requires the := mode for each flight control - channels 5/6 being unused).

It would be good if it were possible to set these modes with a special function (triggered by a switch movement), rather than having to enter the X9E's menu system and change each one of the 4 manually each time the role of the X9E changed.

kilrah, 2018-09-05:
For your head tracker you should not be using the Trainer function at all. If it's on CH5 and 6 then you're using TR5 and TR6 in the mixer and not using the Trainer function.

closed 2018-09-05 by kilrah

geekerator, 2018-09-05:
Thanks Kirah, I am now on the correct road to managing my situation (without your guidance, I would be blundering around for days)...