opentx/opentx #6157: Simulator doesn't force firmware to quit, may hang/crash Companion also.

mpaperno, 2018-09-06:
Right now Simulator assumes that the firmware properly shuts down/exits when requested (at least eventually). 

If the firmware hangs up or prevents shutdown (eg. during checklist display) then closing Simulator or restarting radio will cause Simulator to hang, requiring a manual forced closing.  If launched from Companion this causes it to crash as well.  In the latter case it becomes a bigger deal since it may cause one to loose changes in open file(s).

My suggestion would be to implement some supervisor (timer) process which will force firmware shutdown if it doesn't happen within a set amount of time after the request.  I'm not entirely sure what this means in practical terms (w/out looking at code), probably just killing the hanging thread(s) (assuming that can be done safely). 

I think the user needs to be notified if this happens, as it is still a good way to find FW bugs/issues. 

Personally I've assumed up to this point that anything preventing shutdown of radio IS a bug.  But bug or feature, it doesn't seem nice to whoever "discovers" it if their Companion crashes.

Issue originally pointed out by @rdeanchurch in https://github.com/opentx/opentx/issues/6073#issuecomment-410554381

Somewhat (un)related, it has occurred to me in the past that if Simulator was always launched as a standalone app, instead of being built into Companion, this would avoid any Sim crashes also crashing CPN.  The only real issue to address, I think, is data exchange, but that can happen via a temp. file (or similar).  If anyone has thoughts on this, please share (maybe we start separate thread).

mpaperno, 2018-09-06:
I should add that there is already a mechanism for detecting if firmware hung during normal running, as Simulator expects a heartbeat from the firmware every x ms.  The issue is when the firmware _is_ running normally, just not shutting down when requested.

3djc, 2018-09-06:
As a side comment, that heartbeat process can be an issue when the pc goes on standby : when resuming it detect a failed heartbeat

closed 2019-11-15 by stale[bot]