opentx/opentx #6165: Model name on Top bar

offer-shmuely, 2018-09-10:
I like to request enhancement for the top bar on X10
I like to be able to choose “model name” (same as “model picture”) to be displayed

I know there is an option to put text, and write again the model name manually,
But I do copy models and change the name often for testing, and the name change is not reflected on the screen on that case

 

OpenTx 2.2.2 
X10s
Enhancement

lshems, 2018-09-10:
Send a request at www.justfly.solutions and I'll make you this widget.

offer-shmuely, 2018-09-10:
I feel that displaying the model name is basic feature for all.
Not just me.
Every other transmitter software present the (real model name) on the screen.

kilrah, 2018-09-10:
It is also there by default, just not in the top bar.

kilrah, 2018-09-10:
No, just a choice. Full default widget displays both name and image, was decided the topbar size widget would show image only becasue the dimensions fit better. A long name would come in tiny characters in a single line in the top bar.

[image]

But anyway that's exactly why the Horus UI was made to be fully customizable, so that everyone can do whatever they want.

offer-shmuely, 2018-09-10:
I believe that good approach should be:
Make the common things easy, and the hard things possible.

kilrah, 2018-09-10:
Define common. You're the first to ask for that in 2+ years OpenTX for Horus has existed.

mpaperno, 2018-09-10:
Here you go:
https://github.com/mpaperno/OTX-Lua/tree/master/WIDGETS

[image]

The model image widget can probably be modified to do the same thing (as an option)... But this one you can use now. In the screenshot, the image widget is first in the top bar, then the name widget. Text size/color is adjustable.

offer-shmuely, 2019-05-13:
suggest to close this issue, and accept the pull request that add widget "ModelName"
see: https://github.com/opentx/opentx/pull/6411

closed 2019-11-10 by kilrah