opentx/opentx #6166: Volume - add “min volume”

offer-shmuely, 2018-09-10:
Assignment of pot to volume on special-functions is cool
However some time I set it too low (by mistake) and I don’t hear the voltage alarm that is the trigger for my landing 

I like to add to the volume (on the special-function) a min-volume level, so the pot will change the level between my min-vol-preset to the max level

Enhancement 
X10
OpenTx 2.2.2

3djc, 2018-09-10:
This is not a support place, please go to places like rcgroups or openrc forums for usage questions, as OpenTX already allow you to do that.

closed 2018-09-10 by 3djc

offer-shmuely, 2018-09-10:
Why support??

I request an enhancement that will be useful for other.

Today I do it with other trick, using channels, but I feel that the openTx can be better.
That why I suggest an enhancement.

schwabe, 2018-09-10:
@offer-shmuely because what you want can already be achieved. But this is not the place to explain how.

3djc, 2018-09-10:
There is no need to use channels, you simply set an input how you want it. This is a standard OpenTX mechanism that applies not only to that use, but any other similar use

offer-shmuely, 2018-09-11:
3djc, input is very similar to the channel in that sense of defining 2 lines, one in input+one in special-func x 10 models = 20 lines 
is there more elegant solution?

schwabe, where do you put the max()? I did not find such option

pindeslandes, 2018-09-11:
Maybe it's time to remember that one: #2777

kilrah, 2018-09-11:
> input is very similar to the channel in that sense of defining 2 lines, one in input+one in special-func x 10 models = 20 lines

And? That's only twice as much as if something was added to the special function. But adding something  would mean a few dozens of hours of work for us, a change in eeprom structure, and you having to wait probably for a year to have it in a future major OpenTX version.

3djc, 2018-09-12:
They way we tend to see the 'elegant solution' is by providing basic building blocks, Lego like in a way, that allow you to tailor the radio to your need further than a classic fixed UI based would allow.

Regarding audio only, we have seen tons of questions from people  that wanted to adjust their sound level based on telemetry (f no telem, it means they are at home and don't want to risk waking up the baby), or that are using jets that produce massive sound above 75% throttle, and want their radio louder when they make those full power low pass (but not when the aircarft is far away enough).  Some want to put sound on a 3 pos switch, but then min/max do'nt work, as the middle is unlikely to have the right volume at (max-mmin)/2, and the list goes on.

In short, everybody is different. Trying to put that in 'classic ui choices' is not very nice. Giving you blocks that allow you to do that yourself requires a bit more work, yes, admited,  but allow you to go much further in tailoring to your need. Also, the mechanic to do that is not unique to sound volume, it is the very same lofgic and tools that are used through the radio, so there is nothing special to know/learn to be able to make those complex scenarios.

offer-shmuely, 2018-09-13:
WOW, I never thought about mixing the volume level with throttle or distance, must try it.
Point taken!
Can you refer me to a place that can show discuss how to do the “min vol” that I use on radio-level, and not per model?
with as match lego bricks as needed as long as they  are radio level?
Can I use somehow the radio-level-custom-function with kua script?

3djc, 2018-09-13:
Anything telemetry related will have to be at model level, and cannot be done at radio level, same as anything needing something else than raw input, so in your case, you cannot set it radio wide (well, it would be a specific situation if you could, most/many flyers have a variety of models which all produce different noise level, therefore you minimum is unlikely to be the same for all model)

Radio wide functions are very limited by nature, as they cannot know what will exist when a model is loaded. It can gives you a basic volume control for example, but more evolved things, like settings min/max/curved volume need to be done at model level

As for LUA,from (my getting older) memory I do not think you can control radio volume, but I'll think about that and make some tries

pindeslandes, 2018-09-14:
@offer-shmuely You might be interested in this solution (my radio is an X9D+ but I guess it should be very similar on an X10):
I use a small function script which sets an input line (Input 32, in my case, which obviously must not be used for anything else in any model that uses global functions) and that is called only once (at model loading) by a global function.
This way, any model that uses global function automatically gets these volume settings, making them radio-wide.
A little drawback is you can’t change the range of the volume on the field and need to modify the script for that, but it’s fine for me as I don’t need to change the range once it’s set (from 30% to 75% of the maximum volume of the radio in my case).

The script (sorry, I haven't translated the comments):
_-- VolSon.lua
--[[Fonction : réglage du volume sonore
Insère une ligne Entrée 32 permettant le réglage du son de 30% à 75% du volume maxi de la radio avec RS
Rq : au-dessus de 75%, le volume n'augmente quasiment pas => limitation à 75% ppour optimiser la sensibilité du réglage
A coupler avec une fonction globale FGx	: Volume / [E32]Son
]]--
local parametres={}
parametres.name = 'Son'
parametres.source = 83 -- 83 = 'rs'
parametres.weight = 45
parametres.offset = 5
parametres.switch = 0 -- 0 = aucun
--[[
o	name (string) : input line name
o	source (number) : input source index
o	weight (number) : input weight
o	offset (number) : input offset
o	switch (number) : input switch index
--]]
model.deleteInput(31, 0)
model.insertInput(31, 0, parametres)
return_

The global functions:
[image]