opentx/opentx #6167: Companion not displaying CELS, Cel1 etc telemetry data, other data OK

boblane1949, 2018-09-11:
Companion simulator for Cels, Cel1, Cel2, Cel3, CDif, and LCel telemetry data is not displayed in companion nor do logical switches using them seem to work.  There is no field for them in the simulator.

**To Reproduce**
Steps to reproduce the behavior:
1. Attached file is a copy of otx, and log files
2. Open Companion and open Trex450.otx from zip
3. Select model Trex450 and simulate radio
4. Load the log file in the simulator and run thru it:  Trex450-2018-09-10.csv from zip

**Expected behavior**
The simulator radio telemetry screens 1 and 2 are set to display these and other telemetry data.  Fields like RPM, HEAD, Batt display data - Cels, Cel1, Cel2, Cel3, CDif, and LCel  do not display data nor to logical switches work based on these fields.

**General information**
 - FrSky X9D Plus OpenTX 2.2.2  7-28-2018 release version
 - FrSky X8R, LiPO voltage sensor and RPM & Temperature sensors, 3S battery
 - Windows 10 64 bit all available updates installed
 - OpenTX Companion 2.2.2 July 28, 2018
[OpentTX Cels Trbl.zip](https://github.com/opentx/opentx/files/2368899/OpentTX.Cels.Trbl.zip)

mpaperno, 2018-09-11:
I believe you are referring to the telemetry simulator **log reader** specifically.  The "manual" simulator does have fields for individual cells, and they work.

Most of the fields you mention are "custom" (not saying uncommon). However the log reader does not have support for custom telemetry fields (or even standard ones whose names have been changed). This is a duplicate of #4434 and/or #4256.

As for "Cels", the log reader is both broken (or possibly outdated), but also has a problem. The problem is there is no reliable way to get the individual cell voltages from one log entry of the total. Even if you specify the number of cells somehow (via a new setting), at best it could just average the total out across that many cells.  

Right now it is supposed to take the "Cels" total and use that as the value for the first simulated cell.  But even that part is broken, because it looks for the log header with "Cels(gRe)" instead of "Cels(V)".  I don't know what "gRe" was supposed to be, must be something old.  I will fix this last part and close this, but a true fix would need to be part of addressing the bigger issues (linked in the dupes).

closed 2018-09-11 by mpaperno

boblane1949, 2018-09-11:
Max,

Thank you for confirming what I suspected was going on.  Here is a summary
of what I finally ended up with it.

After re-flashing the radio with the same 2.2.2 bin file used before,
deleting and rediscovering the sensors they came back.  I did need to go in
(directly on the radio)  and manually set the ratios to 1 from the default
of 0.

The LiPo telemetry works well on X9D Plus using OpenTX 2.2.2 July 28 2018.
You do have to set the ratios within the radio from 0 to = 1. (for
Cel1...Cell6, LCel and CDif)
Telemetry does not work in Companion telemetry simulator.
Programming of Logical and Physical Switches in Companion 2.2.2. and up
loading to radio, that refer to the LiPo telemetry data (Cels, Cel1,
etc....) works as expected.

Seems to be just a Companion issue.

Regards, Bob

kilrah, 2018-09-11:
As mentioned it is normal. Your CelX sensors are not real sensors, they are "Calculated" as pointed out by you choosing that when setting them up. 

They are calculated from data that are not written to the log, so cannot be replayed. 
They can be fed by the telemetry simulator, but not replayed from the log.

commit 773b563d (2018-09-11): [Simulator] Fix telemetry simulator log reader expected header for Cels column (closes #6167).
commit 31b9f805 (2018-09-15): [Simulator] Fix telemetry simulator log reader expected header for Cels column (closes #6167).