opentx/opentx #6169: Log file updates when new sensors added

boblane1949, 2018-09-11:
When using logging, the first use of the (radio) day creates a log file with that days date.  This file is appended with each use on that date.  If a new sensor is added, the log file does not add the header names to that days log file, but it does start to log the new data.  This results in a log file with data that has no column header.  Deleting the .csv log file and forcing creation of a new log file will result in all sensors having a column heading.

Maybe an alert option to create a new log file when sensors are added.

FrSky X9D Plus OpenTX 2.2.2 July 28, 2018
FrSky X8R, LiPo and RPM sensors

kilrah, 2018-09-11:
#2542

closed 2018-09-11 by kilrah

boblane1949, 2018-09-12:
Thank you for your time and looking into this.  Sorry I wasted your time
and missed the previous reports of this issue.

Thx Bob