opentx/opentx #6173: Trainer system malfunction - Master tx unable to take back control

TomWEW, 2018-09-17:
Two X7S transmitters were programmed and connected with a jack cable via their respective Trainer ports for use in model aircraft flight training.

10 flights were made, over a period of a week, in the Master/Slave configuration. Control of the aircraft was routinely switched perfectly satisfactorily several times between Master and Slave during each flight. On the next flight, control of the aircraft from take-off was continually with the Slave unit. After approximately nine minutes, the Instructor released the Trainer switch on the Master unit to take control of the aircraft, but control remained with the Slave unit held by the Pupil. Despite several operations of the Trainer switch to and from Slave and Master positions on the Master unit, the aircraft remained continuously under the control of the Slave unit.
The aircraft was safely landed under control of the Slave unit, during which time the Trainer switch on the Master unit was in the Master position.

The aircraft was made safe by removing the flight battery; capacity checks showing that 13% charge remained. Both transmitters had ‘normal’ screens displayed with no alarms or failure notices, so both were switched off.

The jack lead was checked, and the jack plugs were found to be fully inserted into the transmitters, so the Txs were switched back on, a fully charged flight battery connected and the Master/Slave system checked. It functioned correctly and continues to function correctly on the bench in the workshop.

I can find no fault with the jack cable and the Trainer switch also checked out ok.

I'm quite confident that neither the jack cable, nor the Trainer switch are the cause of the malfunction experienced. 
If the cable or Trainer switch were to be faulty, it is reasonable to expect to find an open circuit, or break of continuity, somewhere.
However, the Trainer switch in its 'hands free' normal position (‘Master’ state) has its contacts open. When the toggle is moved against the spring, to the position where it is held to give control to the Slave Tx, the contacts are closed. So, the fault would have to be a short-circuit in the Trainer switch. I have found no evidence of this in the Trainer switch.
During the malfunction, and also when the system is healthy, the Master Tx receives control signal commands from the Slave Tx via the cable, so the cable was neither open-circuited, nor short-circuited when the malfunction occurred.

I suspect that there could be a bug in the Trainer function software.

Could the long time with no activity from the Master unit have caused, erroneously, for control to be locked to the Slave unit?

The Tx assigned to the Master function and the Tx assigned to the Slave function display identical information in the Radio Setup page 5/9:

FW	: opentx-x7
VERS : 2.2.2 (e3ea8595)
DATE ; 2018-06-24
EEPR : 218

3djc, 2018-09-17:
Hard to tell, but we never had similar reports, so an obvious bug is unlikely. Please attach the models otx, because a model error creating those symptoms is possible too, where another switch (logical or physical) could interfere with trainer switch

kilrah, 2018-09-17:
Can you try to repeat the same scenario on the ground?
If it happens again, can you disconnect the cable and see if the master correctly regains control?

TomWEW, 2018-09-17:
I'm a beginner to open-tx, please help me to attach the  models otx, I don't understand the abbreviation otx.

TomWEW, 2018-09-17:
I have found what I have to do to attach the otx file. I will get back to you after downloading the OpenTX Companiaon software

TomWEW, 2018-09-17:
Companion doesn't appear to support otx file type in the Issues threads, so
I have attached the Slave and Master files here.

mpaperno, 2018-09-17:
E-mail attachments are not supported on GitHub either. To post a `.otx` file here just rename it with a different extension which GitHub allows, like `.zip` or `.txt` (or add .zip to the end, eg. myfile.otx.zip).  (.otx files are in fact just Zip-compressed archives.)

closed 2019-11-15 by stale[bot]