opentx/opentx #6176: OpenTx 2.2.2 - LS and RS not found in companionTX and mixer inputs missing

jreis1, 2018-09-18:
Opentx 2.2.2 stable version. 
companionTX for macOS. 

I'm here to report 2 issues, that may be or not related. 

I've a model set on my taranis X9D+ in which LS and RS side buttons are attributed to 2 PWM channels in the mixer.

1) Everything works great, however when importing the model from Taranis to Companion to edit it easily via the software, the LS and RS inputs disappear from the mixer if those inputs are edited and saved.  They are also not on the inputs list.  If not modified in anyway,  those mixers inputs are OK. 

Please attached pictures. 
[image]
[image]
[image]

jreis1, 2018-09-18:
As consequence of the issue reported above, when re-uploading the edited model from the CompanionTX to the radio, some input lines from the mixer disappear. 

Here the File that I've edited in CompanionTX 
[image]

Here what can be seen in Taranis Mixer: (line CH9 to CH11 missing)
[image]

However, if downloading again the same model from the Radio to OpenTx, the lines that are missing on the mixer in the radio are still visible!

kilrah, 2018-11-21:
There is no fix becasue there is no problem, he just likely had wrong settings in the menu I posted about above.

KRCNZ, 2018-11-21:
kilrah, maybe he did, so please advise why my settings have mysteriously changed since I flashed my system with 2.2.2? I have been running RS and LS on my X9D+ for 3 years - and all of a sudden my hardware profile has changed - the only difference is I have flashed it with 2.2.2?

3djc, 2018-11-21:
You are not saying which version you are coming from, but earlier version did not care much about those settings, now it does matter and needs to be right

KRCNZ, 2018-11-21:
3djc - OK that is a thought - thank you. I came from 2.2.1 (opentx-x9d+-lua-luac-en-2.2.1). The version I am running now is 2.2.2 (a5de713b). We have only upgraded because we will be playing around with R9M modules and an X-Lite.

Obviously not a wide spread issue otherwise everyone would have the problem. I have just been into work and we have another 3 x X9D+ there and another here - all on the same version. 2 out of the 5 Tx's have the symptoms - so that makes it a bit trickier...

The reason I have voiced an opinion on this thread is I dont think issues should be closed without some explanation giving a reason as to why - or saying "we dont know - we'll leave it open for a little while to see if anyone else has an issue"...