opentx/opentx #6181: Companion 2.2 MacOS crashes on LUA error or overload

zenuavsolutions, 2018-09-20:
Process:               companion [74513]
Path:                  /Applications/OpenTX Companion 2.2.app/Contents/MacOS/companion
Identifier:            ???
Version:               ??? (???)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           companion [74513]
User ID:               502

Date/Time:             2018-09-20 11:32:13.054 +0200
OS Version:            Mac OS X 10.13.6 (17G65)
Report Version:        12
Anonymous UUID:        9D2EED7A-AC89-1183-8C25-AD1BFD2D8348

Sleep/Wake UUID:       CAA28476-F7B0-4EAB-BC08-B34A633C8458

Time Awake Since Boot: 460000 seconds
Time Since Wake:       13000 seconds

System Integrity Protection: enabled

Crashed Thread:        17

Exception Type:        EXC_BAD_ACCESS (SIGBUS)
Exception Codes:       KERN_PROTECTION_FAILURE at 0x0000000112ee8f2c
Exception Note:        EXC_CORPSE_NOTIFY

VM Regions Near 0x112ee8f2c:
    __LINKEDIT             0000000112e4f000-0000000112e91000 [  264K] r--/rwx SM=COW  /Applications/OpenTX Companion 2.2.app/Contents/Resources/libopentx-x12s-simulator.dylib
--> mapped file            0000000112e91000-0000000112f22000 [  580K] r--/rwx SM=COW  Object_id=d1aae809
    mapped file            0000000112f22000-0000000112fcb000 [  676K] r--/rwx SM=COW  Object_id=d1aaf109

Thread 0:: Dispatch queue: com.apple.main-thread
0   org.qt-project.QtGui          	0x0000000109a8f9f1 QPainterPath::elementCount() const + 1
1   org.qt-project.QtGui          	0x0000000109adb2e3 QTransform::map(QPainterPath const&) const + 83
2   org.qt-project.QtGui          	0x0000000109ada484 QTransform::map(QRegion const&) const + 260
3   org.qt-project.QtWidgets      	0x000000010922d7cb QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) + 3915
4   org.qt-project.QtWidgets      	0x000000010922de38 QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) + 1192
5   org.qt-project.QtWidgets      	0x000000010922dcc6 QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) + 822
6   org.qt-project.QtWidgets      	0x000000010922dcc6 QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) + 822
7   org.qt-project.QtWidgets      	0x000000010922dcc6 QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) + 822
8   org.qt-project.QtWidgets      	0x000000010922dcc6 QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) + 822
9   org.qt-project.QtWidgets      	0x000000010922dcc6 QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) + 822
10  org.qt-project.QtWidgets      	0x000000010922dcc6 QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) + 822
11  org.qt-project.QtWidgets      	0x000000010922dcc6 QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) + 822
12  org.qt-project.QtWidgets      	0x000000010922dcc6 QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) + 822
13  org.qt-project.QtWidgets      	0x000000010922dcc6 QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) + 822
14  org.qt-project.QtWidgets      	0x000000010922dcc6 QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) + 822
15  org.qt-project.QtWidgets      	0x000000010922d8b0 QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) + 4144
16  org.qt-project.QtWidgets      	0x000000010922de38 QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) + 1192
17  org.qt-project.QtWidgets      	0x000000010922d8b0 QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) + 4144
18  org.qt-project.QtWidgets      	0x000000010922de38 QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) + 1192
19  org.qt-project.QtWidgets      	0x000000010922dcc6 QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) + 822
20  org.qt-project.QtWidgets      	0x000000010922d8b0 QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) + 4144
21  org.qt-project.QtWidgets      	0x000000010922de38 QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) + 1192
22  org.qt-project.QtWidgets      	0x000000010922d8b0 QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) + 4144
23  org.qt-project.QtWidgets      	0x000000010922de38 QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) + 1192
24  org.qt-project.QtWidgets      	0x000000010922d8b0 QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) + 4144
25  org.qt-project.QtWidgets      	0x000000010922de38 QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) + 1192
26  org.qt-project.QtWidgets      	0x000000010922d8b0 QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) + 4144
27  org.qt-project.QtWidgets      	0x00000001092012ee 0x1091e2000 + 127726
28  org.qt-project.QtWidgets      	0x0000000109233ffc QWidget::event(QEvent*) + 2716
29  org.qt-project.QtWidgets      	0x000000010934599c QMainWindow::event(QEvent*) + 1660
30  org.qt-project.QtWidgets      	0x00000001091f3a4d QApplicationPrivate::notify_helper(QObject*, QEvent*) + 269
31  org.qt-project.QtWidgets      	0x00000001091f6392 QApplication::notify(QObject*, QEvent*) + 5906
32  org.qt-project.QtCore         	0x0000000109f6e5c4 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 164
33  org.qt-project.QtCore         	0x0000000109f6f1f8 QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) + 600
34  org.qt-project.QtWidgets      	0x0000000109524182 0x1091e2000 + 3416450
35  org.qt-project.QtCore         	0x0000000109f986fc QObject::event(QEvent*) + 156
36  org.qt-project.QtWidgets      	0x00000001095301c3 QGraphicsScene::event(QEvent*) + 1155
[... 525 lines cut ...]
===========                     =======  ======= 
TOTAL                            922.5M      805 

Model: MacBookPro10,1, BootROM MBP101.00F6.B00, 4 processors, Intel Core i7, 2,3 GHz, 16 GB, SMC 2.3f36
Graphics: Intel HD Graphics 4000, Intel HD Graphics 4000, Built-In
Graphics: NVIDIA GeForce GT 650M, NVIDIA GeForce GT 650M, PCIe
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D5434314753364D465238432D50422020
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D5434314753364D465238432D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0xEF), Broadcom BCM43xx 1.0 (7.21.190.18.1a3)
Bluetooth: Version 6.0.7f10, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM256E, 251 GB
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: FaceTime HD Camera (Built-in)
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: Hub
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: USB 3.0 Bus
USB Device: Express Mouse
USB Device: FrSky Horus Mass Storage
Thunderbolt Bus: MacBook Pro, Apple Inc., 23.4

kilrah, 2018-09-20:
Any info about what you were doing at the time maybe?

mpaperno, 2018-09-20:
Yea, need a reproducible case.  Crash appears to be in `DMAFillRect()` which is most likely a firmware issue, not Companion.  But that's pretty much the only useful line in that whole crash dump.

zenuavsolutions, 2018-09-21:
sorry for the long crash dump, i don't know which information is important.
the crash appears when i start Simulator with complex LUA telemetry widget. The widget works on my Horus without any issue. I tried this with companion 2.2.2 and with nightly build. After some crashes every time i started simulation, i replaced the App with a fresh downloaded on and it never appeared. No idea whats happened.

kilrah, 2018-09-21:
You should provide a copy of your widget. Seems it might be doing something wrong (e-g- drawing outside of the screen) that we'd need to ensure is not risky on the radio...

closed 2019-11-15 by stale[bot]