opentx/opentx #6182: X-Lite sounds list bug

FlyerFpv, 2018-09-20:
Hello everyone.

Just received my Xlite and when Im trying choose the music from list, the list just appear with bugs and invisible.
I can select the sounds with joystick but I cant see what Im selecting. 
Anyone with the same issue?
Best regards
[image]

kilrah, 2018-09-20:
Have you updated to latest OpenTX?

FlyerFpv, 2018-09-21:
> 
> 
> Have you updated to latest OpenTX?

I have 2.2.2 218

kilrah, 2018-09-21:
I mean have you updated to OUR latest distribution. We do not support what comes stock from FrSky since it never matches any of releases even when they show the same version number.

FlyerFpv, 2018-09-21:
> 
> 
> I mean have you updated to OUR latest distribution. We do not support what comes stock from FrSky since it never matches any of releases even when they show the same version number.

Solved, already working.
The problem was the different time.
Now I have OpenTX 2.2.2 variant 218 with different time and its working!
Thanks

closed 2018-09-21 by kilrah