opentx/opentx #6186: Incorrect timer countdown (THs) in Spanish Languaje

Luznatural, 2018-09-24:
**Description of the bug**
In all tested radios (already tested in Taranis X9D+, QX7, and Horus X10, with same results) selecting Spanish Languaje as default, every time the countdown (THs) in any timer passes through an exact minute, the obtained result is the voice telling (example) "3 minutos y cero segundos" that translated to English means "3 Minutes and cero seconds" being the "y" spelled, not pronounced.
This issue is also produced with amber voice pack, and some other sound packs, so we assume is not sound-file related, but some flaw in the program that tries to pronounce also the seconds.

**To Reproduce**
To reproduce the problem, you just have to select ES (Spanish) as Default Languaje in the System Menu, and select any plane with a voiced countdown timer (THs). Every time you pass through a minute, the problem is reproduced.

**Expected behavior**
I expect just hearing the voice telling the exact minute (Without added seconds) until the last ten seconds, when you hear the countdown from ten to cero, exactly equal to selecting English Languaje.

**Screenshots**
Not applicable.

**General information**
 - OpenTX version: reproduced at least with versions 2.1.9, 2.2.0, 2.2.1, and 2.2.2
 - Hardware already tested: Frsky X9D+, Horus X10, Frsky QX7.
 - OpenTX Companion (Windows version)

I offer to expand the information, test, and report the proposed solution, at least the radios listed.

fmartinezo, 2018-09-24:
To qualify the described example  by @Luznatural the word seconds is never pronounced, Taranis only says "one minute i Greek zero" for the first minute elapsed and for the following it says "two i Greek  zero", "three i Greek zero", ....

3djc, 2018-09-26:
I'm not fully sure I understand clearly what you are saying, but I have pushed a change anyway. Please test with tomorow's nightly (you can test on simu, not need to flash radio), and let me know if the fix is correct

CBMSOLRAC, 2018-09-26:
The real problem is that the machine that translates the texts into Spanish confuses the letter "y" with the letter "i," and pronounces "y" instead of "i"
parameter 114 of Sounds / is / System / must be moved

fmartinezo, 2018-09-26:
Yes @CBMSOLRAC and the sound for the first minute will be sound 112  (un)  sound 118 (minuto) instead sound 113 (una) sound 118 (minuto).  for the second will be sound 2 (dos) sound 119 (minutos) , for the third will be sound 3 (tres) sound 119 (minutos),.......

CBMSOLRAC, 2018-09-26:
Correct, you're right there are many mistakes in Spanish, a complete revision of sounds ( es) would be necessary

3djc, 2018-09-26:
That requires someone with knowledge of spanish, programming and willing to do that work. So far, nobody has volunteered for that

kilrah, 2018-09-26:
Code is here: https://github.com/opentx/opentx/blob/2.2/radio/src/translations/tts_es.cpp

fmartinezo, 2018-09-26:
As far as I could see, I Think that in the code would have to replace PUSH_NUMBER_PROMPT (ES_PROMPT_UNA) for the case of minutes and seconds (for the hours is ok) by PUSH_NUMBER_PROMPT (ES_PROMPT_UN) and in the sounds change the corresponding to greek "y" by "i" (Latin "i")

fmartinezo, 2018-09-26:
Latin "i" sound as first sound for "eo" from word "people" in english

3djc, 2018-09-27:
How that letter is prounouced is a matter of sound pack and can be easily changed (i will to do it now for our pack). What requires more work is the "grammar" piece of it (when to use plural,...)

fmartinezo, 2018-09-27:
I get it. For the case of time, in Spanish all that is only the unit for minute and second is used the singular with the number defined as "un" instead of "uno" or "una" for example "un minuto", "un segundo" the rest of values is used in plural minutes, seconds, preceded by the number of them, for example "three minutes", "fifteen seconds". In the case of hours (hour is female) "una" is used for the value 1 followed by "hora" and for the rest of values the number and the word "horas" for example "una hora", "tres horas".
For both hours, minutes and seconds when the value is 0 always the time term is plural. for example "cero segundos" And in the case of term "Y" always is pronounced as "i"  (see y.zip)

fmartinezo, 2018-09-27:
I just tried the file 114 modified to "i" and generated with the opentx sonde generator and it is practically inaudible with the voice of "Ms Helena Desktop" and the voice of "Ms Zira Desktop" mispronounces it. He does it like the pronunciation in English

3djc, 2018-09-27:
We only do provide basic sound packs to get people started (they are not generated using the tool you mention). It is up to the community to produce better ones.

CBMSOLRAC, 2018-09-27:
@fmartinezo Test of raising the gain with the "audacity" program, + 5dB for me is correct and works well with Helena in spanish

fmartinezo, 2018-09-27:
Do you know when the grammar corrections we've been talking about will be available?
Here you have the file system 0114 generated for the "Y" correctly pronounced and perfectly audible
[0114.zip](https://github.com/opentx/opentx/files/2425362/0114.zip)

3djc, 2018-09-27:
Our sound pack is automatically generated from the code, we cannot add and mix any specific sound.

As for the change, as long as a Spanish speaking programmer does not volunteer to work on those changes, that’s unlikely to make any progress

fmartinezo, 2018-09-27:
Here you have it

Case hours
       If number of hours = 1
              Use ES_PROMPT_UNA followed by ES_PROMPT_HORA
       Else
             Use number of hours followed by ES_PROMPT_HORAS
      Endif
Case minutes
      If number of minutes = 1
             Use ES_PROMPT_UN followed by ES_PROMPT_MINUTO
      Else
             Use number of minutes followed by ES_PROMPT_MINUTOS
      Endif
Case seconds
        If number of seconds = 1
            Use ES_PROMPT_UN followed by ES_PROMPT_SEGUNDO
        Else
            Use number of seconds followed by ES_PROMPT_SEGUNDOS
        Endif

It is not necessary in Spanish to use the term "Y" when they say hours followed by minutes, minutes followed by seconds or the three followed.

fmartinezo, 2018-09-28:
@3djc It Would be possible to make these modifications and include them in a nightly?

Line 234 replace PUSH_NUMBER_PROMPT (ES_PROMPT_UNA) by PUSH_NUMBER_PROMPT (ES_PROMPT_UN)

Line 256 replace PUSH_NUMBER_PROMPT (ES_PROMPT_UNA) by PUSH_NUMBER_PROMPT (ES_PROMPT_UN)

Delete lines 244,245,246 or comment it. "Y" is not mandatory

fmartinezo, 2018-09-29:
@3djc  Hello, I just modified the part of the time account in the program. I enclose it as a word file, since I can not touch the code. You have to replace the whole code with this one. If you can put it in a Nightly build I try it as soon as you tell me it's done
[voices modif.docx](https://github.com/opentx/opentx/files/2431031/voices.modif.docx)

fmartinezo, 2018-09-29:
@3djc The modifications affect only the last I18N_PLAY_FUNCTION function, which is where hours, minutes and seconds are handled. Please check it out since I did it and I do not know c ++ very well

3djc, 2018-10-03:
I have made changes that are in todays nightly (679). You will need updated sd card content as well (https://downloads.open-tx.org/2.2/nightlies/sdcard/)

fmartinezo, 2018-10-03:
@3djc I have checked the program and these are the results
01:00:00 say "una horaS" must say "una hora"
02:00:00 say "dos"           must say "dos horas"
00:01:00 say "un minutoS" must say "un minuto"
00:02:00 say "dos"            must say "dos minutos"
02:03:00 say "dos tres"     must say "dos horas tres minutos"
02:01:00 say "dos un minutos"  must say "dos horas un minuto"
01:01:00 say "una horas un minutos" must say "una hora un minuto"

3djc, 2018-10-04:
There seems to be an issue with the sd pack, looking into it

3djc, 2018-10-04:
Ok, the server is having a lot of difficulties building those packs recently, but it is done. please test with sdcard version 00018  04-Oct-2018 07:12 found here : https://downloads.open-tx.org/2.2/nightlies/sdcard/

fmartinezo, 2018-10-04:
@3djc Tested it with sd card 00018 04/10/18 and it speak great, all my test was sucesfully
02:02:00 say "dos horas dos minutos" OK
02:01:00 say "dos horas un minuto" OK
02:00:00 say "dos horas" OK
00:02:00 say "dos minutos" OK
00:01:00 say "un minuto" OK
and the countdown is also OK
from 00:0059 to 00:00:31 is mute OK
00:00:30 say "treinta segundos" OK
from 00:00:29 to 00:00:21 is mute OK
00:00:20 say "veinte segundos" OK
from 00:00:19 to 00:00:11 is mute OK
00:00:10 say "diez" OK
00:00:09 say "nueve" OK
......... OK
00:00:02 say "dos" OK
00:00:01 say "uno" and beep OK
For me its all OK great work. Thanks so much. Best regards

3djc, 2018-10-04:
Thanks for your tests 👍

closed 2018-10-04 by 3djc

fmartinezo, 2019-01-08:
Hi @3djc I just downloaded the latest version 2.2.3 of open tx and the version of the SD. I see in the version of the sd that is the 2.2V0018.zip of the 28-Jul-2018 for taranis x9 and 2.2V0018.zip of the 28-Jul-2018 for the horus X10. It has not been corrected therefore the error that we had seen in this issue, since the 0114.wav file is still the old one and continues counting badly

3djc, 2019-01-08:
True, I have updated, would you mind checking ?

fmartinezo, 2019-01-08:
I just installed version 2.2.3 on my horus, I tried the countdown and it says 1 minutes instead of 1 minute

fmartinezo, 2019-01-08:
I'm going to pass the test I did before ( 04/0ct/2018) and I'll tell you

fmartinezo, 2019-01-08:
Checked on opentx for taranis x9d+ all is ok

fmartinezo, 2019-01-08:
I think that the version for horus x10 2.2.3 has not been updated as if it was done for the x9d +

3djc, 2019-01-09:
I just checked here on x10 simu and it is ok, Can you make sure to redownload the horus pack and retry ?

fmartinezo, 2019-01-09:
Redownload it and now works. Thanks so much