opentx/opentx #6192: cannot install in ubuntu

mchaos, 2018-09-29:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
Cannot install 2.2.3 in linux ubuntu, install fails

**To Reproduce**
Steps to reproduce the behavior:
CLI
sudo dpkg -i companion22_2.2.3N678_amd64.deb
ERROR:
dpkg: dependency problems prevent configuration of companion22:
 companion22 depends on libqt5core5a (>= 5.7.0); however:
  Version of libqt5core5a:amd64 on system is 5.5.1+dfsg-16ubuntu7.5.
 companion22 depends on libqt5multimedia5 (>= 5.6.0~beta); however:
  Version of libqt5multimedia5:amd64 on system is 5.5.1-4ubuntu2.
 companion22 depends on libqt5svg5 (>= 5.6.0~beta); however:
  Version of libqt5svg5:amd64 on system is 5.5.1-2build1.
 companion22 depends on libqt5widgets5 (>= 5.7.0); however:
  Version of libqt5widgets5:amd64 on system is 5.5.1+dfsg-16ubuntu7.5.
**Expected behavior**
A clear and concise description of what you expected to happen.
it to install 
**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**

 - OpenTX Companion (e.g. 2.2.3 on Ubuntu 16.04)

kilrah, 2018-09-29:
Not a problem with us, it means your distro doesn't provide a Qt version recent enough for what companion needs aka is too old.

closed 2018-09-29 by kilrah

3djc, 2018-09-29:
Yeah hard to do on 16.04, but 18 works out of the box